nion-software/nionswift #463: Certain combinations of number of pixels and display width make log-line plots disappear

Brow71189, 2020-06-04:
I couldn't figure out what exactly triggers this, but one way to reproduce the issue is to acquire a spectrum with the usim eels camera, switch the display to log scale and zoom out on the x-axis. At some point the spectrum diappears.
Without log scale I never saw the problem.

Interestingly for example `show(numpy.random.rand(1024))` does not show the problem.

cmeyer, 2020-06-05:
This might be happening without log display too. Doing live 1D acquisition, the thumbnail flickers with data and no-data.

cmeyer, 2021-04-17:
The line plot calculates left and right data channels, but checks whether they are within range of the data. If not, the drawing is skipped.

https://github.com/nion-software/nionswift/blob/8bd826436ecb26fe00e4644e09e23752a8f9f30a/nion/swift/LineGraphCanvasItem.py#L354-L365

Also see 8bd826436ecb26fe00e4644e09e23752a8f9f30a.

cmeyer, 2021-05-25:
Closing this until a specific example is found.

closed 2021-05-25 by cmeyer