nion-software/nionswift #470: Sequence Measure shifts produces transposed data

ndellby, 2020-06-23:
processing/sequence/measureshifts outputs its results transposed for 2D shifts, as a sequence N of 2 pixel lines, when it should be two (x,y) lines of length N, so that the default display of 1D data can display the data without further processing.

cmeyer, 2020-06-24:
The reason I chose N x 2 is the the N portion can represent the sequence/collection dimensions. So if you are measuring shifts on a  collection [n x m] that the result is [n x m x 2]. Also, we store complex and RGB data is the fastest changing index (the last index). It seems awkward to produce a [2 x n x m] in the case of more collection dimensions.

In the bigger picture, I'm considering whether our data type system needs to be expanded so that the we can store vectors and other data natively. We already do this (partially) with RGB and complex; so extending data to allow storage of vectors (2d, 3d, etc.) is not a huge stretch. Standardizing the storage format for vector fields/arrays would help settle questions like this.

If we supported vectors as a data type, then the line plot and raster could have the option of plotting the data directly.

I also considered using a complex value for the result.

Perhaps a short term work around is to specially handle this case in the line plot.

ndellby, 2020-06-24:
It seems to me that the problem is more in how the line plot handles 
data than in the format.

For RGB and Complex and now shift images, each pixel has a 
muti-dimensional value, if I have an RBG image x,y,rgb and I draw a line 
profile I would expect to get a X,RGB data set, if I display that as a 
line profile, I would expect three lines red green and blue, with a 
length equal to the span in x. And isn't an SI just x,y,E where E spans 
the spectrum length?

To put some perspective onto it, with the Xray data, and once we get 
some nice EELS analysis going the EELS data as well, then we will have 
the last item be a list of elements like in Sr-Ti-O, i.e, just like an 
RGB but with an arbitrary number of items. So yes a vector valued data 
type where the "dimension" of the vector can be qualitative rather than 
quantitative would be the proper way of doing this.

Complex is a paste on, and line profiles don't really have the right 
behavior for complex anyway.