nion-software/nionswift #475: Swift UI font scaling sometimes scales the wrong way

ndellby, 2020-07-09:
e.g., setting font scaling to 150% makes names in titlebars of panes smaller
[image]

cmeyer, 2021-05-24:
Unfortunately, on Windows the only way to ensure proper font behavior in Swift after changing screen resolution is a reboot. If a problem persists after reboot, please file a new issue with details or re-open this one.

ndellby, 2021-05-25:
re-boot windows or re-start Swift? One is a major mess the other very minor.

Niklas

cmeyer, 2021-05-25:
Unfortunately, re-boot Windows.

I'm still looking into this to see if there is some way to detect the changed behavior by only restarting Swift. I was just about to re-open this.

cmeyer, 2021-05-25:
This is much improved now - no reboot required - just application relaunch.

This problem comes down to Qt mishandling the resolution changes for default widget styling. To solve, I added explicit font-size for all widgets (via the stylesheets), display panel headers, and a few other spots. Also Qt 5.9 (conda), Qt 5.12 (conda-forge) and Qt 5.15 (nionui-tool) have minor differences in behavior. There are also a few more problem fonts remaining (fast/slow button in scan control panel), but I'm closing this issue as resolved - file issues for anything remaining.

See the following commits:

b5bdc3e4f64f24b7f3340bdd550a17bf4623e1bf
nion-software/nionui-tool@b3dc74f0859eea6a272d60d4c7e9b59a47c49cc6
nion-software/nionui-tool@ffec7c6b6912a873d59a40d69091a277c02ca31d
nion-software/nionui@72245e61738c6a5cb02bdf9f527bfbf741aff42c
nion-software/nionui@52a917bd1977cace2917dc8bcb2e3bdfbeb1945d

closed 2021-05-25 by cmeyer