nion-software/nionswift #488: Version 2 projects should be upgraded (silently or with confirmation) to version 3

cmeyer, 2020-07-23:
Version 3 does not include connections - so writing to a version 2 project will make it incompatible with older versions of Swift.

ndellby, 2020-07-23:
Remember, all old obsolete version projects must be readable by all 
future versions, and tests must be added.

Phil has more than once pulled up APL projects with data and 
calculations from 30 years ago and just run them.

If upgrading old stuff breaks old analysis that's a problem.

cmeyer, 2020-07-23:
@ndellby This version loads all previous versions. There are a total of 16 file versions -- 13 "library" versions and then renamed to "project" and now 3 project versions). There are tests for all versions.

commit 1129a2e5 (2020-07-27): Fix #488. Require v2 projects to be upgraded to v3.
closed 2020-07-27 by cmeyer