nion-software/nionswift #492: Switching projects should show progress bar when loading new project

cmeyer, 2020-07-28:
Currently, any existing window is closed and the new project loads without giving any feedback to the user. This can be confusing for the duration of the load, which can be several seconds for large projects.

MattRoyle, 2026-09-07:
I would like implement this as a follow up to project renaming #1927. The gap where the current project is closed and renamed then reopened can take several seconds which can easily be confused for failure. 
Does this need a progress bar or could it just be a splash screen to reduce complexity?
Many applications have a progress message like "Building shaders" or "loading project data", this could be the last message in the terminal.

cmeyer, 2026-09-07:
A modal dialog would be a great start.

MattRoyle, 2026-09-10:
We discussed in backlog refinement the notes are below:
This could be expand to “Loading projects” although in the build version there is already a splash screen.
Potentially show last log message as a indicator of progress.
Having a *throbber* (animated loading icon) probably wont work even if it updated on a separate thread. It also wouldn't be as helpful as a progress bar in terms of informing the user as to the progress of switching/loading.
Might be useful to have stages of loading, i.e. loading plugins, reading data files, creating display items etc.
Loading data items may be a place that has more granular progress to show loaded out of total items.
Also show current data item being loaded so we know which it has failed on
SPIKE: check we can do anything here other than 0 & 100% for a progress bar.

This could be done in stages of having the initial modal splash screen before moving on to implementing user feedback of loading/progress.