nion-software/nionswift #494: User should be given choice to load existing known projects when switching/upgrading fails

cmeyer, 2020-07-28:
If a project cannot be loaded, the current version shows the open project file dialog. It would be better to offer the user to either open another known project (recent project) or open another one on disk or create a new one.

closed 2020-07-29 by cmeyer

AEljarrat, 2020-07-29:
Hi Chris, 

thank you for looking into this issue. I've been able to open nionswift after the fix but there were still some issues:

- When opening nionswift a lot of error messages appeared in the console. I was prompted to select or create a project file. I created one and then nionswift opened.

- After this, I tried to open the project I just created created but could not find it. I then tried to create a new project, but more errors appeared in the terminal.

You can see these error messages pasted at the end of this post. It seemed to me the errors were related with nionswift not finding camera sources and such. Following this logic I installed the `nionswift-usim` package. After this the error messages subsided. I am currently able to open nionswift and create/load projects without errors :)

Best regards,
Alberto

### 1/2 Opening nionswift ###
(base) aeljarrat@AG-SEMs-MBP ~ % ~/opt/miniconda3/envs/nionswift_dev/bin/nionswift
Using Python from command line: /Users/aeljarrat/opt/miniconda3/envs/nionswift_dev
Launch time 2020-07-29 16:09:04.123405
Python version 3.7.7 (default, May  6 2020, 05:01:30) [Clang 4.0.1 (tags/RELEASE_401/final)]
User interface class QtUserInterface / HostLibProxy
Qt version 5.14.1
Application data: /Users/aeljarrat/Library/Preferences/Nion/Nion Swift/nionswift_appdata.json
Loading plug-ins from /Users/aeljarrat/Library/Application Support/Nion/Nion Swift/PlugIns
NOT Loading plug-ins from /Users/aeljarrat/Documents/Nion/Swift/PlugIns (missing)
Plug-in 'nionswift_plugin.none' loaded (nionswift_plugin).
Plug-in 'nionswift_plugin.DM_IO' loaded (nionswift_plugin).
Plug-in 'nionswift_plugin.TIFF_IO' loaded (nionswift_plugin).
Video device configuration: /Users/aeljarrat/Library/Preferences/Nion/Nion Swift/video_device_config.json
Plug-in 'nionswift_plugin.nion_instrumentation_ui' loaded (nionswift_plugin).
Plug-in 'nionswift_plugin.nionswift_video_capture' NOT loaded. Could not import cv2. (nionswift_plugin)
Plug-in 'nionswift_plugin.nion_eels_analysis' loaded (nionswift_plugin).
Plug-in 'nionswift_plugin.nion_experimental_4dtools' loaded (nionswift_plugin).
Plug-in 'nionswift_plugin.nion_experimental_tools' loaded (nionswift_plugin).
NOT Loading color maps from /Users/aeljarrat/Library/Preferences/Nion/Nion Swift/Color Maps (missing)
Using existing profile /Users/aeljarrat/Library/Application Support/Nion/Nion Swift/Profile.nsproj
Using cache /Users/aeljarrat/Library/Application Support/Nion/Nion Swift/Profile Cache.nscache
Loading project /Users/aeljarrat/Documents/Nion Projects/Nion Swift Project 20200729.nsproj
Exception creating panel 'MultiAcquire-Panel': 'NoneType' object has no attribute 'scan_controller'
Traceback (most recent call last):
File "/Users/aeljarrat/NionApps/nionswift/nion/swift/Workspace.py", line 1045, in create_panel_content
    panel = cls(document_controller, panel_id, properties)
File "/Users/aeljarrat/NionApps/nionswift/nion/swift/Facade.py", line 3166, in create_facade_panel
    panel.widget = panel_delegate.create_panel_widget(ui, document_controller)._widget
File "/Users/aeljarrat/NionApps/nionswift-instrumentation-kit/nionswift_plugin/nion_instrumentation_ui/MultiAcquirePanel.py", line 635, in create_panel_widget
    self.multi_acquire_controller.superscan = self.superscan
File "/Users/aeljarrat/NionApps/nionswift-instrumentation-kit/nionswift_plugin/nion_instrumentation_ui/MultiAcquirePanel.py", line 152, in superscan
    self._superscan = self.stem_controller.scan_controller
AttributeError: 'NoneType' object has no attribute 'scan_controller'
File "/Users/aeljarrat/NionApps/nionswift/nion/swift/Application.py", line 196, in start
    document_controller = self.open_project_window(project_reference)
File "/Users/aeljarrat/NionApps/nionswift/nion/swift/Application.py", line 226, in open_project_window
    document_controller = self.create_document_controller(document_model, "library")
File "/Users/aeljarrat/NionApps/nionswift/nion/swift/Application.py", line 315, in create_document_controller
    document_controller = DocumentController.DocumentController(self.ui, document_model, workspace_id=workspace_id, app=self)
File "/Users/aeljarrat/NionApps/nionswift/nion/swift/DocumentController.py", line 141, in __init__
    self.__workspace_controller = Workspace.Workspace(self, workspace_id)
File "/Users/aeljarrat/NionApps/nionswift/nion/swift/Workspace.py", line 91, in __init__
    self.create_panels(visible_panels)
File "/Users/aeljarrat/NionApps/nionswift/nion/swift/Workspace.py", line 164, in create_panels
    dock_panel = self.create_panel(document_controller, panel_id, title, positions, position, properties)
File "/Users/aeljarrat/NionApps/nionswift/nion/swift/Workspace.py", line 173, in create_panel
    panel = self.workspace_manager.create_panel_content(document_controller, panel_id, title, positions, position, properties)
File "/Users/aeljarrat/NionApps/nionswift/nion/swift/Workspace.py", line 1052, in create_panel_content
    traceback.print_stack()
Exception creating panel 'MultiAcquire-Panel': panel is None [MultiAcquire-Panel]
Traceback (most recent call last):
File "/Users/aeljarrat/NionApps/nionswift/nion/swift/Workspace.py", line 174, in create_panel
    assert panel is not None, "panel is None [%s]" % panel_id
AssertionError: panel is None [MultiAcquire-Panel]
File "/Users/aeljarrat/NionApps/nionswift/nion/swift/Application.py", line 196, in start
    document_controller = self.open_project_window(project_reference)
[... cut ...]
    traceback.print_stack()
Exception in callback Window.request_close()
handle: <Handle Window.request_close()>
Traceback (most recent call last):
  File "/Users/aeljarrat/opt/miniconda3/envs/nionswift_dev/lib/python3.7/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/Users/aeljarrat/NionApps/nionui/nion/ui/Window.py", line 289, in request_close
    self.__document_window.request_close()
AttributeError: 'NoneType' object has no attribute 'request_close'

cmeyer, 2020-07-29:
Thanks for the additional info. I'm tracking the secondary bug here:
https://github.com/nion-software/nionswift-instrumentation-kit/issues/35

commit bbe46b35 (2020-07-29): Fix #494. Allow user to choose recent, create new, or open existing projects.