nion-software/nionswift #497: Baseline in line plot log plots is treated strangely

cmeyer, 2020-07-30:
`show(numpy.array([3,3,4,4,7,1,3,3,4,4]))`

Now drag the vertical axis so that it is constant.

Now drag the horizontal axis to be centered on the 3rd column.

Now expand the horizontal axis around the 3rd column. The baseline jumps around. This seems counterintuitive - although it is just choosing the lowest displayed value.

How does Matplotlib do it? Any thoughts?

[image]

[image]

Brow71189, 2020-08-03:
I see 2 issues here:
1) Some bug is preventing the data to be drawn all the way to the left edge of the plot. I'll try and track that one down.
2) There is no "natural" baseline for a log-axis like there is for a linear axis. Right now the implementation uses 1e0 (i.e. 1) as a baselinge, but I'm not sure that this is the right choice. I can also see 0 as a baseline, which would mean that the filled area reaches down infinitely when you scroll towards smaller y-values.

I didn't manage to get the same example running in matplotlib because their filled line plot does not fill from the baseline but from the minimum of the data being drawn (which looks even more awkward btw.)

Do you have any preference on the baseline in log mode?

cmeyer, 2020-08-03:
For the natural baseline, I would look at how other programs do it: Wolfram Alpha or Mathematica, Origin, MATLAB, or Excel.

commit 745593e5 (2020-08-04): Always use x-axis as baseline for log-plots (consistent with other plotting software). Fixes #497.
closed 2020-08-04 by cmeyer