nion-software/nionswift #516: Add ability to designate dependent/source data to be used in displays, computations

cmeyer, 2020-08-17:
For example, if background subtraction is applied to a line plot, the user could process the original data (re-binning or smoothing, for example) and then designate the resulting data to be used anywhere the original data is used (computations, displays, maybe more).

Other ideas are: if the designated replacement is deleted, it reverts to the original source; some specialized operations may have the option to auto-designate without explicitly doing so (UI needed).

ndellby, 2020-08-17:
What about cloning, if i have processed one set of data in 10 steps, and i
want to do the same thing on another set as well?

cmeyer, 2020-08-17:
@ndellby I added #517 Clone processing to track that. Thanks.