nion-software/nionswift #519: Enable auto-chunking on HDF5 files

cmeyer, 2020-08-20:


Brow71189, 2020-08-20:
Do you mean auto-chunking like the actual h5py `chunks=True` option or some chunking based on our data layout (i.e. SI or 4D data)?

cmeyer, 2020-08-20:
Good point. Let's start with chunks=True ... but maybe we should switch things around so that instead of passing around `large_format` to various creation methods, we pass `data_descriptor`. Then the low level IO can have more control over how it manages chunking. I'll remove the "easy" tag now.

cmeyer, 2020-08-20:
As a side note, we might want to continue performance tests before proceeding with this too much further. We need to answer the question: Is datum-shape chunking really better then auto?

commit 826fa79f (2020-08-27): Fix #519. Use chunking for large HDF5-files. Add tests.
closed 2020-08-27 by cmeyer