nion-software/nionswift #527: h5py copies non-contiguous data which can cause excessive memory usage

Brow71189, 2020-08-27:
This is most likely the reason why we are only able to handle datasets that are well below the installed memory of a PC. The implementation of `__setitem__` in h5py (which gets called when you do things like `dataset[some_slice] = data[some_other_slice]`) copies data if it is not c-contiguous.
This happens here: https://github.com/h5py/h5py/blob/e5b1806b81fa15133e6be69cdbeb22be1677180b/h5py/_hl/dataset.py#L864

For example slicing along any axis but the first one makes a numpy array non-contiguous (e.g. when removing flyback pixels).

I'm not sure we can avoid non-contigous data but I just wanted to make sure we are aware of where the extra memory usage is coming from.

cmeyer, 2020-08-27:
Is there an issue that we can fix here?

cmeyer, 2020-08-27:
Also, I think there are two properties to be aware of: 'contiguous' and 'ordering'. Ordering can be 'C' (last index is fastest moving index in memory) vs 'F' (Fortran: first index is fastest moving). Contiguous is different... it is whether the strides (`data.strides`) match the size of the data (dimension size x the dtype size). In some sense, they're similar in that 'F' ordering is not considered to be 'contiguous' when viewed from the 'C' ordering perspective. Slicing for flyback keeps the data in 'C' order, but makes it non-contiguous. I'm wondering if there is some confusion in the NumPy docs about this too. The phrase "C continuous" might be confusing the two concepts.

Brow71189, 2020-08-28:
Well in this context `order = 'C'` seems to mean "c-contiguous". You can see that from the following code:

```python
import numpy as np
a = np.ones((4, 4))
b = a[:-1]
c = a[:, :-1]
print(b.flags)
print(c.flags)
b2 = np.asarray(b, order='C')
c2 = np.asarray(c, order='C')
print(b2.flags)
print(c2.flags)
```

Compare the array flags printed: `b` and `b2` have `C_CONTIGUOUS : True`  and `OWNDATA : False`. `c` has `C_CONTIGUOUS : False`  and `OWNDATA : False` and `c2` has `C_CONTIGUOUS : True`  and `OWNDATA : True`, which means that a copy was made from `c` whereas `b2` is simply a view of `b`.

Answering you first question:

> 
> 
> Is there an issue that we can fix here?

No, at least not for the general case. But we should keep in mind that this is happening and maybe we can try to keep data c-contiguous in some situations, especially when working with large arrays.

ndellby, 2020-08-28:
Would it help to make sure that flyback data gets removed in the torrent 
reception so that there are no flyback gaps?

Niklas

cmeyer, 2020-08-28:
I'm closing this issue since there is no action to be taken.

https://github.com/nion-software/nionswift-instrumentation-kit/issues/42 is the response to @ndellby's question.

closed 2020-08-28 by cmeyer