nion-software/nionswift #529: 1D-Collection of 1D-Data does not update display when chaning "Index" slider in Inspector

Brow71189, 2020-09-07:
To reproduce, open a console and run:

```python
show(numpy.random.rand(2, 32))
```
Then go to `Processing -> Redimension Data ->Redimension to Collection of 2 Spectra of Length 32`

Try changing the "Index" slider in the Inspector to see the second "slice" of data. The display does not update.

This is with current master branch of nionswift.

cmeyer, 2020-09-18:
Interestingly, sequence works; collection doesn't.

cmeyer, 2020-09-18:
Also, `show(numpy.random.rand(2, 32))` works. This is because there is a special case to allow for small collections (< 16) of 1D data to display as a layered line plot. Adding a 'planning' tag.

cmeyer, 2020-09-18:
```
index af28f39..2a61bb0 100755
--- a/nion/data/Core.py
+++ b/nion/data/Core.py
@@ -1882,7 +1882,7 @@ def function_element_data_no_copy(data_and_metadata: DataAndMetadata.DataAndMeta
         collection_dimension_count = data_and_metadata.collection_dimension_count
         datum_dimension_count = data_and_metadata.datum_dimension_count
         # next dimensions are treated as collection indexes.
-        if collection_dimension_count == 1 and datum_dimension_count == 1 and data_and_metadata.collection_dimension_shape[0] <= 16:
+        if False and collection_dimension_count == 1 and datum_dimension_count == 1 and data_and_metadata.collection_dimension_shape[0] <= 16:
             pass  # this is a special case to display a few rows all at once. once true multi-data displays are available, remove this
         elif collection_dimension_count == 2 and datum_dimension_count == 1:
             data_and_metadata = function_slice_sum(data_and_metadata, slice_center, slice_width)
```

commit fa43dd2f (2020-12-04): Fix #529. Treat collection of 1D as collection, not image.
closed 2020-12-04 by cmeyer