nion-software/nionswift #543: PyPI and conda-nion tool builds are broken on Linux

cmeyer, 2020-09-25:


PhilippPelz, 2020-10-15:
Hi Chris, just ran

on my Ubuntu 16.04

conda create -n nionswift -c conda-forge nionswift nionswift-tool
conda activate nionswift

then I ran

nionswift 

and got

 /home/philipp/anaconda3/envs/nionswift/bin/NionSwiftLauncher/NionSwiftLauncher: /usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5' not found (required by /home/philipp/anaconda3/envs/nionswift/bin/NionSwiftLauncher/NionSwiftLauncher)
/home/philipp/anaconda3/envs/nionswift/bin/NionSwiftLauncher/NionSwiftLauncher: /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5: version `Qt_5' not found (required by /home/philipp/anaconda3/envs/nionswift/bin/NionSwiftLauncher/NionSwiftLauncher)
/home/philipp/anaconda3/envs/nionswift/bin/NionSwiftLauncher/NionSwiftLauncher: /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5: version `Qt_5' not found (required by /home/philipp/anaconda3/envs/nionswift/bin/NionSwiftLauncher/NionSwiftLauncher)

my 

LD_LIBRARY_PATH is

/usr/local/lib:/usr/local/lib64:/opt/arrayfire/lib:/usr/local/cuda/lib64:/usr/local/cuda/nvvm/lib64:/opt/openblas/lib/:/usr/lib/x86_64-linux-gnu/:/usr/local/cuda-8.0/lib64:/usr/local/cuda-8.0/nvvm/lib64:/usr/local/lib:/usr/local/lib64:/usr/local/lib:/usr/local/lib64:/opt/arrayfire/lib:/usr/local/cuda/lib64:/usr/local/cuda/nvvm/lib64:/opt/openblas/lib/:/usr/lib/x86_64-linux-gnu/:/usr/local/cuda-8.0/lib64:/usr/local/cuda-8.0/nvvm/lib64

so it must be picking up the wrong Qt5 library

which Qt5 library are you linking against? Then I can add it to my LD_LIBRARY_PATH as priority.

cmeyer, 2020-10-15:
It _should_ link to Qt in the miniconda installation. It uses rpath's (in the executable) to find the location of Qt (or at least it should). And you shouldn't have to mess with LD_LIBRARY_PATH. I'll have some follow-up questions soon after I see if I can reproduce this locally.

Btw - going forward, weren't planning on supporting Ubuntu 16 much longer - it's official end of life is Apr 2021 according to Ubuntu website. I will take another look at building for Ubuntu 16. I don't remember the specific issue that made it difficult.

PhilippPelz, 2020-10-15:
Now when I link it to the anaconda version of Qt5, I get a little further:

(base) philipp@philipp:~/anaconda3/envs/nionswift/bin/NionSwiftLauncher$ ldd NionSwiftLauncher 
        linux-vdso.so.1 =>  (0x00007ffd30d8d000)                                                                                                                                                                                           
        libQt5Widgets.so.5 => /home/philipp/anaconda3/lib/libQt5Widgets.so.5 (0x00007ff3368b3000)                                                                                                                                          
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007ff3366af000)                                                                                                                                                                
        libQt5Gui.so.5 => /home/philipp/anaconda3/lib/libQt5Gui.so.5 (0x00007ff3362e3000)                                                                                                                                                  
        libQt5Core.so.5 => /home/philipp/anaconda3/lib/libQt5Core.so.5 (0x00007ff335ecc000)                                                                                                                                                
        libstdc++.so.6 => /home/philipp/anaconda3/envs/nionswift/bin/NionSwiftLauncher/./../../lib/libstdc++.so.6 (0x00007ff335d57000)                                                                                                     
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007ff335a4e000)                                                                                                                                                                  
        libgcc_s.so.1 => /home/philipp/anaconda3/envs/nionswift/bin/NionSwiftLauncher/./../../lib/libgcc_s.so.1 (0x00007ff336f2d000)                                                                                                       
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ff335684000)                                                                                                                                                                  
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007ff335467000)                                                                                                                                                      
        libGL.so.1 => /usr/lib/nvidia-418/libGL.so.1 (0x00007ff335128000)                                                                                                                                                                  
        /lib64/ld-linux-x86-64.so.2 (0x00007ff336de5000)                                                                                                                                                                                   
        libpng16.so.16 => /home/philipp/anaconda3/lib/./libpng16.so.16 (0x00007ff336ef3000)                                                                                                                                                
        libz.so.1 => /home/philipp/anaconda3/lib/./libz.so.1 (0x00007ff336ed2000)                                                                                                                                                          
        libicui18n.so.58 => /home/philipp/anaconda3/lib/./libicui18n.so.58 (0x00007ff334ea9000)                                                                                                                                            
        libicuuc.so.58 => /home/philipp/anaconda3/lib/./libicuuc.so.58 (0x00007ff334cf6000)                                                                                                                                                
        libicudata.so.58 => /home/philipp/anaconda3/lib/./libicudata.so.58 (0x00007ff3333f4000)                                                                                                                                            
        libgthread-2.0.so.0 => /home/philipp/anaconda3/lib/./libgthread-2.0.so.0 (0x00007ff336ecc000)                                                                                                                                      
        libglib-2.0.so.0 => /home/philipp/anaconda3/lib/./libglib-2.0.so.0 (0x00007ff3332cb000)                                                                                                                                            
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007ff3330c3000)                                                                                                                                                                
        libnvidia-tls.so.418.40.04 => /usr/lib/nvidia-418/tls/libnvidia-tls.so.418.40.04 (0x00007ff332ebf000)                                                                                                                              
        libnvidia-glcore.so.418.40.04 => /usr/lib/nvidia-418/libnvidia-glcore.so.418.40.04 (0x00007ff33123f000)                                                                                                                            
[... cut ...]               
(base) philipp@philipp:~/anaconda3/envs/nionswift/bin/NionSwiftLauncher$ ./NionSwiftLauncher                                                                                                                                               
./NionSwiftLauncher: relocation error: ./NionSwiftLauncher: symbol _ZN17QAbstractItemView11eventFilterEP7QObjectP6QEvent, version Qt_5 not defined in file libQt5Widgets.so.5 with link time reference      

Which exact version of Qt5 does Swift need?

PhilippPelz, 2020-10-15:
It should really be compiled against the anaconda Qt5 library, then it just works out of the box.
How can I recompile it from source?

PhilippPelz, 2020-10-15:
Just tested on Ubuntu 18.10. I get the same error message:

/mnt/data/software/anaconda/envs/nionswift/bin/NionSwiftLauncher/NionSwiftLauncher: relocation error: /mnt/data/software/anaconda/envs/nionswift/bin/NionSwiftLauncher/NionSwiftLauncher: symbol _ZN17QAbstractItemView11eventFilterEP7QObjectP6QEvent version Qt_5 not defined in file libQt5Widgets.so.5 with link time reference

cmeyer, 2020-10-15:
Ok, if you're using conda-forge, you will need to run `conda install -c conda-forge qt` to install the conda-forge version of Qt (it goes into the site-packages and should not interfere with other installations of Qt).

PhilippPelz, 2020-10-15:
Works like a charm. Thank you for the quick response and resolution!

cmeyer, 2020-10-17:
0.4.7

closed 2020-10-17 by cmeyer