nion-software/nionswift #551: Allow metadata to be copied from the Metadata Panel

Brow71189, 2020-10-12:
Maybe add a Button "Copy to Clipboard" like in the Task Panel.

cmeyer, 2020-10-15:
What would be the format? JSON? XML? Both?

ndellby, 2020-10-26:
There's a test.

Paste into Powerpoint, Excel and Word, and make sure it comes out 
usable. And also into metadata so one can reattach info that got dropped.

Powerpoint and word are where people seem to put their notes on data, 
and if they are doing an experiment that needs tracking A vs B then it's 
likely to go into excel, copying and pasting the same thing 20 
times-tedious and if they have to jump through hoops for every 
paste-awful. Anyone who is writing actual code to use the data is likely 
to be able to parse anything sensible.

Brow71189, 2020-10-27:
> 
> There's a test. Paste into Powerpoint, Excel and Word, and make sure it comes out usable. 

I agree, but this is actually nontrivial for hierarchical data like our metadata. A table (like in Excel) is not going to be able to represent this type of data properly.

> And also into metadata so one can reattach info that got dropped. 

That is a very good point and would be extremely useful since we loose all metadata when we process data (see also nion-software/niondata#16).

Brow71189, 2020-10-27:
Actually after playing with this for a bit I found out that Excel can parse xml files. Word should also be able to do so but haven't manage to get that work yet.
I've created an xml file from one of our image's metadata that you can just drag-and-drop into excel to see how that looks (after unzipping it).

[image_dict_0.zip](https://github.com/nion-software/nionswift/files/5444658/image_dict_0.zip)