nion-software/nionswift #552: Allow Fourier masks to be offset when applied to non-Hermitian data

cmeyer, 2020-10-14:
This PR was an initial attempt at this: https://github.com/nion-software/nionswift/pull/354

Required for 4D STEM.

Brow71189, 2020-10-22:
We also need new mask types for non-Hermitian data, see nion-software/nion-internal#17.

TomaSusi, 2022-08-16:
I just ran into this issue looking at 4D data (or well, actually 5D time sequence) from the Berlin ELA. The calibration of axis 3 (corresponding to the vertical direction in the displaced ronchigram) was not set in the data, so the ring filter was placed at the first line and could not be moved.

I was very confused for a while what was going on, especially about the lacking possibility to simply drag the mask to the desired location. I asked @Brow71189, and he explained the effect of the missing calibration on the vertical positioning, but even after correcting for this, I noted that the bright-field disk is not exactly centered on the camera and needs to be corrected.

While it my _in theory_ be correct to treat Fourier masks as always centered, _in practice_ real data is not ideal and for Swift to be useful for 4D-STEM analysis, there **must** be an easy way for the user to move this mask to create virtual detectors.

ndellby, 2022-08-17:
In practice Fourier masks are centered, simply because the bottom half 
is just a hermitian conjugate of the top. But a circle that is aligned 
with an image does not live in Fourier space.

So fixing the center of the mask on real data is not even right /in theory./

Niklas

cmeyer, 2022-08-17:
For FFT's resulting from real data, he Fourier masks are centered on the calibrated origin, which is 0.5, 0.5 pixels.

You can change the calibrated origin to anything you like and the Fourier mask will be centered on that origin.

If you take your image to which you're applying the mask, you can edit the x/y calibrations. Note that the offset is in units of fractional coordinates; the default for FFT is such that taking the `pixel * scale + offset` gives a fractional coordinate equal to the 0.5, 0.5 pixel location. Also note that there is a [#689 bug](#689) that if you change the calibration, the masks do not redraw. Overall this isn't a great user experience.

For your specific case, can you change the calibrated origin on your data until we have a better solution?

TomaSusi, 2022-08-17:
I do agree that when the data item _really_ is an FFT, then offsetting the center doesn't make sense. However, the ring masks are extremely useful for making virtual annular dark field detectors for 4D-STEM data, where the data item isn't an FFT and thus the center isn't mathematically fixed.

And yes, I did manage to work around the issue by changing the scaling offsets, but I think that clearly for 4D-STEM data making it possible to drag the mask would be ideal.

cmeyer, 2024-04-15:
Closing as resolved. See also
- https://github.com/nion-software/nionswift/issues/362

closed 2024-04-15 by cmeyer