nion-software/nionswift #566: Fourier masks are calculated wrong for non-square data

Brow71189, 2020-11-12:
Like in the example linked below, all Fourier masks use the size of the first axis for both dimensions in ``get_mask()``:

https://github.com/nion-software/nionswift/blob/136330597be61915bd005b919662fed9f6bce78b/nion/swift/model/Graphics.py#L1699

Brow71189, 2020-11-12:
#567 Fixes the center issue, but the ring mask is still wrong for non square data. To reproduce create a ring mask on non-square Fourier data and run the following on the command line:

``show(r07.graphics[0]._graphic.get_mask(r07.data.shape))``

This is what you get from it:

[image]

I'm not entirely sure what the intended behavior is here. Should the graphic be shown as a circle on the data instead of an ellipse or should the mask also be an ellipse like the graphic implies?

cmeyer, 2020-11-12:
Generally, the graphic aspect ratio for Fourier masks should be the same as the calibrated aspect ratio of a single pixel (i.e. the calibration scale x over calibration scale y) and the data should match the graphic.

cmeyer, 2025-04-09:
Try the PR and let's see how it turns out. Maybe this requires #109 first?
- #109

cmeyer, 2025-04-09:
Also, what about just passing the aspect ratio to the drawing code? I don't think drawing needs calibrations (at least yet).