nion-software/nionswift #568: Error from updating 0.14.8 to 0.15.1

agust2010, 2020-11-13:
After updating from version 0.14.8 any spectrum image that has been reduced prior to the update with a "pick (sum)" or "pick (average)" has an error if the "display bounds" are moved. 

The anaconda terminal continually repeats the following error message, freezing the software. 

```
:
'NoneType' object is not subscriptable
Traceback (most recent call last):
File "C:/Users/Agust/Miniconda3/envs/nionswift/Scripts/NionSwiftLauncher\bootstrap.py", line 25, in bootstrap_dispatch
return getattr(object, method_name)(*args)
File "C:\Users\Agust\Miniconda3\envs\nionswift\lib\site-packages\nion\ui\QtUserInterface.py", line 1862, in periodic
self._handle_periodic()
File "C:\Users\Agust\Miniconda3\envs\nionswift\lib\site-packages\nion\ui\UserInterface.py", line 2927, in _handle_periodic
self.root_widget.periodic()
File "C:\Users\Agust\Miniconda3\envs\nionswift\lib\site-packages\nion\ui\UserInterface.py", line 572, in periodic
child.periodic()
File "C:\Users\Agust\Miniconda3\envs\nionswift\lib\site-packages\nion\ui\UserInterface.py", line 572, in periodic
child.periodic()
File "C:\Users\Agust\Miniconda3\envs\nionswift\lib\site-packages\nion\ui\UserInterface.py", line 572, in periodic
child.periodic()
File "C:\Users\Agust\Miniconda3\envs\nionswift\lib\site-packages\nion\ui\UserInterface.py", line 2206, in periodic
self.on_mouse_position_changed(*self.position_info)
File "C:\Users\Agust\Miniconda3\envs\nionswift\lib\site-packages\nion\ui\CanvasItem.py", line 3011, in __mouse_position_changed
self.__mouse_tracking_canvas_item.mouse_position_changed(canvas_item_point.x, canvas_item_point.y, modifiers)
File "C:\Users\Agust\Miniconda3\envs\nionswift\lib\site-packages\nion\swift\LinePlotCanvasItem.py", line 709, in mouse_position_changed
self.__update_cursor_info()
File "C:\Users\Agust\Miniconda3\envs\nionswift\lib\site-packages\nion\swift\LinePlotCanvasItem.py", line 1022, in __update_cursor_info
line_graph_canvas_item = self.line_graph_canvas_item
File "C:\Users\Agust\Miniconda3\envs\nionswift\lib\site-packages\nion\swift\LinePlotCanvasItem.py", line 254, in line_graph_canvas_item
return self.__line_graph_stack.canvas_items[0] if self.__line_graph_stack else None
TypeError
:
```

cmeyer, 2020-11-13:
I could not reproduce this problem. I created a Swift 14 project. Imported SI data. Did both a Pick and a Pick Region. Then I switched to Swift 15 (using version 0.15.2, not yet released) and opened the older project. Swift 15 upgraded the project. I dragged the SI data and the pick regions into display panels and dragged the display bounds around. No errors.

After upgrading, if you restart Swift, does the problem fix itself?

agust2010, 2020-11-13:
It does not fix the problem after restarting swift. I found out that the problem happens only when using the mouse to change the bounds, it does not happen when changing "start" and "end" in the inspector. I've tried restarting swift, as well as the computer overall.

cmeyer, 2020-11-13:
It looks like this is occurring in a composite line plot, which is a line plot display that shows multiple 1D data items.

Is the Swift 14 project small enough to upload to DropBox or something similar? If so, I can download it and try to reproduce the issue that way. You may also be able to copy the original project, open the copy in Swift 14, delete all unrelated data, and then upload the reduced copy. However, if you're doing that, I do not suggest trying to import the copy into Swift 15 as Swift 15 will notice that the copy has already been upgraded and use that instead.

If either of these are possible, you will have to email me directly with the download info. I won't post my email info here, so if that's a problem, leave another message here and I'll figure out a way to get you my email without putting it on a public forum!

cmeyer, 2020-11-13:
I have duplicated (but not explained) the error. I will include a fix in the 0.15.2 release, which should be available later today or by Monday.

agust2010, 2020-11-13:
Thank you, if there continues to be a problem I can get into contact with you and send you the original .nslib where the problem originated.

cmeyer, 2020-11-14:
In the end, I'm not sure if I fixed your specific problem or not. When you have the chance, can you try updating to Swift 0.15.2 and see if that fixes this issue?

cmeyer, 2020-11-20:
@agust2010 Can you check this with Swift 0.15.2, released last weekend?

agust2010, 2020-11-20:
Yes, I have not tried it yet but will see if the fix works early next week.
Thank you.

agust2010, 2020-12-01:
Sorry for the delay... I've tried updating to the most recent version of nionswift and I am still having the same issue. I'll contact you directly to share a link where you can download the original .nslib

cmeyer, 2021-02-14:
Closing due to inactivity. Please reopen if you're still having this issue. I'd like to fix it if you can send me the original library file.

closed 2021-02-14 by cmeyer