nion-software/nionswift #586: Export batch with no boxes checked overwrites same file repeatedly

cmeyer, 2020-12-15:


cmeyer, 2021-08-16:
XY reports "If I export multiple data with the same name (e.g. NionCCD-1010), only one file can be saved. I can check "Include Sequence Number" to allow them all to be saved. However, could you make this batch export smarter to automatically add sequence numbers if several items use the same file name."

cmeyer, 2021-11-15:
The UI for the batch export resides in [`ExportDialog.py`](https://github.com/nion-software/nionswift/blob/master/nion/swift/ExportDialog.py).

Once the user has configured the options, the [`do_export`](https://github.com/nion-software/nionswift/blob/34bc02b87347cb0f939a546f9d42679ae2c549bd/nion/swift/ExportDialog.py#L159) method loops through the selected display items and exports them to a file using [write_display_item_with_writer](https://github.com/nion-software/nionswift/blob/34bc02b87347cb0f939a546f9d42679ae2c549bd/nion/swift/ExportDialog.py#L185).

cmeyer, 2024-06-12:
Fixed 907bf563f6798a01d31c5848861dbded08428c9d

closed 2024-06-12 by cmeyer

Ion-e, 2024-07-15:
Issue verified as working correctly.

Thanks!