nion-software/nionswift #602: Add a system to manage "hidden" data

Brow71189, 2021-02-02:
There are situations where we want to store data on disk that is associated with some visible data (i.e. a `DataItem` in Swift) but that data should be hidden from the user. An example would be some cached full data set for a `DataItem` that only contains reduced data. Since the data is hidden from the user, it needs to be managed automatically in the background.
Here are the requirements for such a system as I see them today. This is just the "API" for it, not which parts of it will be exposed to the user through a UI:

1. The system should allow any file or folder to be designated as a "managed hidden resource". In particular this means it should not be limited to certain file types.
2. It must provide access to the file or folder on disk.
3. For creating new files or folders the system should allow to request a path where to store them.
4. For each "group" of hidden files there should be a way to specify how many to keep or how long to keep them or how much space the files are allowed to use.
5. There should be a way to "lock" files so that they will not be cleaned up even if they fullfill one of the criteria in 4.
6. It probably needs a locking mechanism so that the system does not try to delete a file while it is used.
7. There should be a way to connect "hidden" files to a visible item in Swift such as a `DataItem`.
8. Maybe we could have a way to transform a "hidden" file into a visible item. Since the hidden files can be of arbitrary format this requires passing a reader to the system that tells it how to interpret the file on disk and provide metadata to be added.
9. Maybe we want an option to select whether and/or which files to include the hidden files in the Swift library so they get copied when the used copies the Swift library.

ndellby, 2021-02-02:
This is a resource that has a 1:1 correspondence with a particular 
library, so the cleanest thing would probably be to have a "Library 
auxiliary data" folder making it obvious which library it is connected with.

Also please do not call things hidden, it leads the mind to the windows 
insanity of hiding folders that contain data that one wants to keep.

Niklas

cmeyer, 2021-03-25:
See also #663 expand data item.