nion-software/nionswift #604: Allow user to quickly open non-default project at launch

cmeyer, 2021-02-03:
Holding option/alt while launching should bring up the open project file dialog.

Launching this way should probably _not_ reset the default project.

A similar option will be to pass `--project /a/b/c.nsproj` on the command line.

See also #479 allow multiple projects to be open.

See also #502 launch by double clicking on project files.

Origin: [Matrix Chat Conversation](https://matrix.to/#/!KRXUOTgiHVgZBIdnuI:matrix.org/$donXlakns4KLGUw5aUJEa73vFK9b3I2xpXMefmpJ1E4?via=matrix.org)

closed 2021-02-22 by cmeyer

tcpekin, 2021-02-23:
Hi @cmeyer, does this now work with Windows when holding down shift before opening nionswift from a shortcut?

cmeyer, 2021-02-23:
Yes, but it requires the latest trunk versions of all libraries and nionui-tool 0.4.9 or later, which isn't quite ready for release.

This feature will be available the next minor release which I anticipate to be at the beginning of March.

commit 043bcea8 (2021-02-22): Fix #604. Bypass opening default project with shift key.