nion-software/nionswift #615: User Guide is empty

jondo, 2021-02-19:
https://nionswift.readthedocs.io/en/latest/user_guide.html has got a chapter structure, but no content in the chapters yet. (I have not used Nion Swift yet, so I cannot contribute.)

icbicket, 2021-03-12:
I'm interested in helping to fill out the user guide documentation, as I get to know my way around the software. How can I best contribute?

cmeyer, 2021-03-12:
@icbicket I started a [Wiki page describing contributing to user documentation](https://github.com/nion-software/nionswift/wiki/User-Documentation). I'd give it another week for me to get the rest of my thoughts written onto that page - but it's still worthwhile reading right now.

I was mostly thinking that I would spend a few days (soon) writing brief documentation since (as the developer) I know the ins-and-outs well. Once that is complete, then I would really like review and updates to the docs from outside contributors.

As an example of the initial brief documentation, it would be somewhat like this, which I wrote last week to coincide with the new features in nionswift 0.15.5: [documentation on workspaces](https://nionswift.readthedocs.io/en/latest/user_interface.html#workspaces). It's quite brief, but is useful as a starting point.

At any time, too, [filing issues](https://github.com/nion-software/nionswift/issues) documenting confusing aspects of the software would be useful.

icbicket, 2021-03-15:
Ok, I may be able to contribute some small sections for some of the basic parts, but I'll let you make a start and I'd be happy to review or add more detail to the docs as needed.

cmeyer, 2022-01-11:
Closing with 24f92e0c11be3119c002380f61a6da01237d6bf1. First pass of docs are complete.

closed 2022-01-11 by cmeyer