nion-software/nionswift #621: Revisit handling of pick-like tools with regards to graphics or masks

cmeyer, 2021-02-25:
The pick/pick-sum/pick-average tools add a graphic and then link that graphic (point or rectangle) to the resulting line plot, summing/averaging in the case of pick-sum/pick-average.

The mapped-sum/mapped-average tools work on the default mask (which can be constructed from various graphics) and produce a 2D collection mapping of the sum/average.

The point graphic does not produce any mask.

In the absence of any graphic, the entire 2d data is assumed to be used.

Both variations have their pro's and con's, but I think the two cases should be consistent.

ndellby, 2021-03-26:
The elegant way to deal with sum versus average is to have two 
simultaneous calibrations counts and counts/pixel, easily selectable in 
the graph.

cmeyer, 2024-12-31:
This issue is inscrutable. Closing. I will file other issues as they arise to work on the linking of graphics to computations/displays.

closed 2024-12-31 by cmeyer