nion-software/nionswift #635: Deadlock when opening project with many large hdf5 items in it

Brow71189, 2021-03-18:
While working with large, hdf5-backed datasets in Swift you generally expierience a lot of UI freezes because IO is done on the main thread (or at least locks the main thread). This also happens when using a conda-forge installation with h5py 3.x.

I zipped the project in question which will hopefully help with finding the issue: https://github.com/nion-software/nion-internal/issues/21

All nion software is installed as developer installation using the latest master branch code.
Here are the packages installed in the conda environment of question:
```
# Name                    Version                   Build  Channel
ca-certificates           2020.12.5            h5b45459_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
certifi                   2020.12.5        py38haa244fe_1    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
h5py                      3.1.0           nompi_py38h022eade_100    conda-forge
hdf5                      1.10.6          nompi_h5268f04_1114    conda-forge
imageio                   2.9.0                      py_0    conda-forge
intel-openmp              2020.3             h57928b3_311    conda-forge
jpeg                      9d                   h8ffe710_0    conda-forge
krb5                      1.17.2               hbae68bd_0    conda-forge
lcms2                     2.12                 h2a16943_0    conda-forge
libblas                   3.9.0                     8_mkl    conda-forge
libcblas                  3.9.0                     8_mkl    conda-forge
libcurl                   7.75.0               hf1763fc_0    conda-forge
liblapack                 3.9.0                     8_mkl    conda-forge
libpng                    1.6.37               h1d00b33_2    conda-forge
libssh2                   1.9.0                h680486a_6    conda-forge
libtiff                   4.2.0                hc10be44_0    conda-forge
lz4-c                     1.9.3                h8ffe710_0    conda-forge
m2w64-gcc-libgfortran     5.3.0                         6    conda-forge
m2w64-gcc-libs            5.3.0                         7    conda-forge
m2w64-gcc-libs-core       5.3.0                         7    conda-forge
m2w64-gmp                 6.1.0                         2    conda-forge
m2w64-libwinpthread-git   5.0.0.4634.697f757               2    conda-forge
mkl                       2020.4             hb70f87d_311    conda-forge
msys2-conda-epoch         20160418                      1    conda-forge
nionswift-tool            0.4.9            py38hd481e31_0    conda-forge
numpy                     1.20.1           py38h0cc643e_0    conda-forge
olefile                   0.46               pyh9f0ad1d_1    conda-forge
openssl                   1.1.1j               h8ffe710_0    conda-forge
pillow                    8.1.2            py38h9273828_0    conda-forge
pip                       21.0.1             pyhd8ed1ab_0    conda-forge
pyreadline                2.1             py38haa244fe_1003    conda-forge
python                    3.8.8           h7840368_0_cpython    conda-forge
python_abi                3.8                      1_cp38    conda-forge
pytz                      2021.1             pyhd8ed1ab_0    conda-forge
scipy                     1.6.0            py38h5f893b4_0    conda-forge
setuptools                49.6.0           py38haa244fe_3    conda-forge
sqlite                    3.34.0               h8ffe710_0    conda-forge
tk                        8.6.10               h8ffe710_1    conda-forge
tzlocal                   2.1                pyh9f0ad1d_0    conda-forge
vc                        14.2                 hb210afc_4    conda-forge
vs2015_runtime            14.28.29325          h5e1d092_4    conda-forge
wheel                     0.36.2             pyhd3deb0d_0    conda-forge
wincertstore              0.2             py38haa244fe_1006    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
zlib                      1.2.11            h62dcd97_1010    conda-forge
zstd                      1.4.9                h6255e5f_0    conda-forge
```

Brow71189, 2021-03-18:
It seems like removing the "workspaces" list from the *.nsproj file fixes the problem. To me this looks like having many large files "open" in a workspace at the same time can cause the deadlock.

cmeyer, 2025-03-20:
@Brow71189 is this still an issue?

Brow71189, 2025-03-21:
I cannot test it with the project linked here, because Swift just tells me "Unable to open project". I haven't seen this problem in a while, but it would be nice to test with the Project that was known to cause a problem.

cmeyer, 2025-03-21:
I was able to download the project file, unzip it, and load it on macOS without any issues.