nion-software/nionswift #639: Metadata focus may be broken

cmeyer, 2021-03-24:
BPS: There is a bug with the metadata viewer. The metadata only appears if you click on the data item in the data panel. Then, if you click anywhere in the metadata panel, the metadata panel goes blank, and only appears if you again click on the data item in the data panel. If you click on a data item in a workspace panel, the metadata does not appear.

CM: I could not reproduce this on macOS. Needs test on Windows.

ndellby, 2021-03-24:
I tried on APM, doesn't reproduce, but metadata goes blank on line 
profiles, so different bug.

cmeyer, 2021-05-25:
Focus handling was updated a few months ago - it might have fixed this issue. Closing until we see it again.

closed 2021-05-25 by cmeyer