nion-software/nionswift #650: Add spreadsheet-style view of raster data

cmeyer, 2021-03-24:


Brow71189, 2026-09-01:
Related but still separate issue: https://github.com/nion-software/nionswift/issues/433

MattRoyle, 2026-09-02:
> Related but still separate issue: [#433](https://github.com/nion-software/nionswift/issues/433)

There is overlap with that issue, this issue may be useful as a sub-issue for 433 if the issue of tables in swift is accepted. 
@Tiomat85 has worked on a UI for tables but this issue seems to be specific to a view on a display item and should be view only.
Potentially being able to normalize the table values or display them in scientific notation could be a useful feature so very skewed values are readable
We should gather info from users and stake holders about the how, the level of ability to modify, etc.
There was some discussion on a maximum number of cells displayed at once, for instance 64x64, as at a certain scale the cells will become unreadable. @Brow71189 suggested something similar to maps, when you zoom out you group waypoints. Something similar could be done to average the cells at a certain scale.
@Ion-e suggested having a separate tab to view the values as spreadsheet instead of having it in a display panel in the workspace by right clicking to show this display panel as spreadsheet that you can zoom in/out of.