nion-software/nionswift #666: Make it easier to find floating panels that are obscured by other panels or on another screen

cmeyer, 2021-03-25:
Current technique involves searching and then closing/opening again. This is difficult on remote connections where only part of the screen is shown.

Maybe holding down option/alt would bring up a small centered window allowing you to close, open, reset position, adjust properties, etc.? It's a hidden option, so not ideal.

Brow71189, 2021-03-26:
Another option would be to make the command that created the panel that you're interested in bring this panel to the foreground and to the current cursor location (or the center of the screen or whatever we decide is a good place).
Because by pressing for example Ctrl + K again the user clearly signalizes that he is interested in the built-in console and doesn't see it on his screen. Otherwise he wouldn't press Ctrl + K but just use it. Same goes for settings dialogs etc.

ndellby, 2021-03-26:
How do you do this for the menu where selecting something alternatingly 
hides and shows, do we say: If you hide and then show within 3s then 
you're obviously hunting?

Brow71189, 2021-04-01:
For a menu that is being toggled between hidden and visible it just keeps doing that. So if you are in the "show" part of the cycle it simply is shown. The worst that could happen is that you have to click the "hide/show" button twice to make your menu appear on the screen.

ndellby, 2021-04-01:
But if the "show" part of the screen isn't visible then it goes between 
hidden and invisible, which is what started this issue.