nion-software/nionswift #669: Allow user to choose which axes of a data hypercube are displayed

Brow71189, 2021-04-01:
Right now we have some hardcoded defaults which make sense for most use cases. But especially for graphically selecting data slices for processing it would be very useful to be able to change the displayed axes.
For instance cropping multidimensional data with a graphical selection is currently only possible along the displayed axes. Being able to change the displayed axes would make this example much more straightforward to implement and use.

cmeyer, 2021-04-01:
This is also required for "data item 2".

cmeyer, 2021-04-01:
The way I'm planning to implement this is by generalizing the "display pipeline" which was partially implemented earlier when working on the brightness/contrast/gamma changes.

The idea is that there are different stages to go from arbitrary data to display data. Roughly, the display pipeline works like this, with options/UI at each stage (taken from `DisplayValues` class):

1. start with raw data
2. extract the raw data element using sequence and/or collection indexes and/or slices if required
3. produce display data by applying complex to real conversion if required
4. apply adjustments by normalizing display data and performing adjustment operation
5. calculate rgb display data by scaling display range and applying color table

In this case, we want to provide options and UI for step 2.

Also see my personal notes from 2020-09-14.

Also see #554 separate processing data chain for related ideas.

ndellby, 2021-04-01:
That's a bit terse, stage 2 can at present be, either a crop and sum in 
one or more dimensions--which I think is what you mean, or multiplying 
some dimensions by a weighting function-which is how the virtual 
detectors work, and how all linear SI works (which in my humble opinion 
is the only kind worth using).

Stage 5 becomes interesting if  2-3-4 support data with more than 1 
dimension.