nion-software/nionswift #677: Accessing display_xdata on a DataItem that is part of a multi-line-plot-display fails

Brow71189, 2021-04-14:
To reproduce, take a 1D data item and drag another one on top of it to create a multi-line-plot data item. Then assign an r-number and try the following:

```python
r04.data_items[0].display_xdata
```

This will result in the following error:

```
Traceback (most recent call last):
  File "<console>", line 1, in <module>
  File "D:\git\nionswift\nion\swift\Facade.py", line 1065, in display_xdata
    return display_data_channel.get_calculated_display_values().display_data_and_metadata
AttributeError: 'NoneType' object has no attribute 'get_calculated_display_values'
```

cmeyer, 2021-04-14:
TL;DR: Use `r04.data_items[0].xdata` instead.

Longer explanation:

The `display_xdata` of `data_item` is a hack for backwards compatibility.

In recent versions of Swift (0.14+), there are separate concepts of _display_ and _data item_. A display is one specific visual representation of a data item. In fact, a single display may represent multiple data items, as in a line plot with multiple layers. So a particular data item may have one or more displays associated with it.

A data item contains low level data, but not display data. So methods which directly access low level data, such as `data`, `xdata`, and `data_and_metadata` are always valid on data items (see TL/DR above). However, there are a number of other methods, provided for backwards compatibility, that are only valid if a single display can be unambiguously associated with the data item. These methods are `display_xdata`, `mask_xdata`, as well as several other deprecated methods like `add_rectangle_region`.

In Swift 15, the script references (e.g. `r04`) are assigned to _displays_ and, to be backwards compatible with older scripts, there are accessor methods such as `display_xdata` which are valid when the display represents a single data item. To support displays with multiple data, you can get a list of the data items with the `data_items` method; however, the data items in a line plot with multiple layers will typically have multiple displays (the line plot with multiple layers and a single display for the data item). So those data items do not have a single display associated with them; and thus, methods like `display_xdata` will fail.

So what's the right solution to this? The problem right now is that there needs to be a `get_display_xdata_for_data_item` method on a `Display` which allows you to pass in the data item and returns the `display_xdata`.

A hack for this right now, which will most definitely not work in Swift 16, is:

`r04._display_item.get_display_data_channel_for_data_item(r04.data_items[0]._data_item). get_calculated_display_values().display_data_and_metadata`

Or, just use `r04.data_items[0].xdata`, which is probably what you want anyway.

Brow71189, 2021-04-14:
Thanks for the detailed explanation. It would still be nice if it would be possible to access the display xdata also for multip-displays, so the issue as you changed it now would definitely be a useful addition.
In the case were I encountered this I am actually looking for the display xdata because the computation that needs it should work also for sequences/collections of 1D data, but initially it needs to know something about the display xdata to set up the UI correctly.
I understand that just doing `r04.display_xdata` cannot work in this case, which is why I accessed a specific data item with `r04.data_items[0].display_xdata`. It would be great if this would work because it would be consistent with how it works for single displays. This could just call `get_display_xdata_for_data_item` internally so that we have both functions available.

cmeyer, 2021-04-14:
.. yes I see how extracting a "datum" from a collection would be a good use case.

The workaround hack `r04._display_item.get_display_data_channel_for_data_item(r04.data_items[0]._data_item). get_calculated_display_values().display_data_and_metadata` should work fine at least until Swift 16.

Brow71189, 2021-04-14:
Ok then I'll maybe use the workaround and add a test so that we will see it failing in time.

closed 2025-03-20 by cmeyer