nion-software/nionswift #679: Thumbnail caching is stored in profile cache and has no garbage collection

cmeyer, 2021-04-16:
Rethink where thumbnails are stored. They should probably be stored with the project, either in a sibling, in `.nsproj` file, or in the primary data directory.

cmeyer, 2021-10-15:
From email today:

> The problem is the cache file is shared by any project - so you can delete the cache file and create a new project, but once you open a large project, Swift will try to recreate all of its thumbnails and store them in the cache file. Then returning to the new project will still have the negative effects of the large cache file.
> 
> Can you try this for now? Create a new Swift project or switch to a smaller one. Exit Swift. Delete the cache file. Relaunch Swift. This procedure should guarantee good behavior until you load a large project again. You can follow this same procedure if it reoccurs.
>

kociak, 2021-10-17:
the "started event/thread" things seems unrelated after few tests

commit 0dc24ff8 (2022-08-30): Fix #679. Enable separate cache file for each project. Purge old ones.
closed 2022-08-30 by cmeyer