nion-software/nionswift #727: Creating display_data_item in Swift does not properly store calibrations

cgadre, 2021-09-14:
can use both bummy data and real dm3/4 data.

```
#from nion.typeshed import API_1_0
from nion.swift.model import ImportExportManager
import numpy as np
import hypserspy.api as hs

path = ""
name = "EELS Spectrum Image.dm4"

def createDummyData():
    #creating dummy data
    data3D = np.random.randn((100, 100, 2048))
    cals = [[0, 1], [0, 1], [-100, 0.5]]
    units = ["nm", "nm", "meV"]
    
    return data3D, cals, units

def getDMdata():
    try:
        DMdata = hs.load(path + "\\" + name)
    except (FileNotFoundError):
        print("File not Found. Make sure path and filename are correct.")
    
    data3D = DMdata.data
    cal = DMdata.axes_manager
    del(DMdata)
    
    cals = np.array([np.array([cal[i].offset, cal[i].scale]) for i in range(len(dims))])
    units = [cal[i].units for i in range(len(dims))]
        
    return data3D, cals, units

IEM = ImportExportManager.ImportExportManager()
api = api_broker.get_api(version="1.0", ui_version="1.0")
ui = api.application._proxy._application.ui

window = api.application.document_windows[0]

#data3D, dims, cals, units = createDummyData()
data3D, cals, units = getDMdata()

data_item = api.library.create_data_item_from_data(3Ddata)

intensity_calibration = api.create_calibration(offset = 0.0, scale = 1.0, units = 'counts')

dimensional_calibration_0 = api.create_calibration(cals[0, 0], cals[0, 1], units[0])
dimensional_calibration_1 = api.create_calibration(cals[1, 0], cals[1, 1], units[1])
dimensional_calibration_2 = api.create_calibration(cals[2, 0], cals[2, 1], units[2])

dimensional_calibrations = [dimensional_calibration_0, dimensional_calibration_1, dimensional_calibration_2]

data_item.set_intensity_calibration(intensity_calibration)
data_item.set_dimensional_calibrations(dimensional_calibrations)

display_panel = window.display_data_item(data_item)
```

cmeyer, 2021-09-14:
Did you mean to close this? Reopening.

Can you provide an example that does not use external libraries and runs when copy and pasted into a Nion Swift script console? `3Ddata` variable is not defined; and `cals[x, y]` is only valid on numpy arrays and does not work with the values returned from `createDummyData`.

cmeyer, 2021-09-14:
```
data_item = api.library.create_data_item_from_data(numpy.random.randn(4, 4, 128))

intensity_calibration = api.create_calibration(offset = 0.0, scale = 1.0, units = 'counts')

dimensional_calibration_0 = api.create_calibration(1, 2, 'd1')
dimensional_calibration_1 = api.create_calibration(11, 22, 'd2')
dimensional_calibration_2 = api.create_calibration(111, 222, 'd3')

dimensional_calibrations = [dimensional_calibration_0, dimensional_calibration_1, dimensional_calibration_2]

data_item.set_intensity_calibration(intensity_calibration)
data_item.set_dimensional_calibrations(dimensional_calibrations)

window = api.application.document_windows[0] 
display_panel = window.display_data_item(data_item)
```

closed 2025-03-20 by cmeyer