nion-software/nionswift #735: Provide a way for plugins to add entries to sub-menus

Brow71189, 2021-09-28:
Currently, ``api.create_menu_item`` only allows you to add entries to or create new top-level menus. In order for menu entries created by plugins to integrate in our "hierarchy" of existing menu options, it would be useful to also add entries to sub-menus from plugins.
This would be particularly important for additions to the "Processing" menu where we use a lot of sub-menus to group similar computations.

cmeyer, 2025-03-20:
See also
- https://github.com/nion-software/nionswift/issues/836

cmeyer, 2025-03-20:
Generally I'd like to discourage users from adding anything but simple commands to menus. For processing, there will be a different system of registering available computations and searching for them. The menu hierarchy is badly organized already and not sustainable as the number of processing items grows.