nion-software/nionswift #763: Allow line width (stroke width) of graphics to be changed

Brow71189, 2021-11-15:
Right now this is hardcoded to "1", but it would be nice if this could be changed.

Brow71189, 2021-11-15:
I got an email from a user asking if this was possible so I thought it would be good to track the request here.

Of course it would be amazing if we had more flexibility in our graphics, so maybe we shoudl change this issue to a more general one.

cmeyer, 2021-11-15:
See also #764.

jamesrussell216, 2024-09-10:
Just adding another use case on to this - it would be very helpful if you could increase (for example) the line thickness as the graphics would be easier to see over VNC link! I think this would help for alignment activities.

cmeyer, 2024-11-26:
PR #1267

Ion-e, 2025-03-28:
https://github.com/nion-software/nionswift/pull/1268

Current PR addressing some of the tasks

Ion-e, 2025-03-31:
Looks good to me, all graphics have been tested except the following: https://github.com/nion-software/nion-internal/issues/203#issuecomment-2761738354

cmeyer, 2025-06-13:
Closed with PR https://github.com/nion-software/nionswift/pull/1268

Follow ups:
- #1559
- #1560
- #1561
- #1562

closed 2025-06-13 by cmeyer

Ion-e, 2025-07-09:
Waiting for confirmation that all specs have been introduced.

Ion-e, 2025-07-18:
Everything is a pass except:
if both stroke and fill are None, stroke should fall back to default color

Actual behaviour: if you type None in the box it doesn't rever to default. Do we want to fix this @lisham2000