nion-software/nionswift #768: Interoperability issue with hyperspy and numba

kociak, 2021-11-28:
The current version of nionswift (0.16.0) is not compatible with the current version of hyperspy (1.7.0.dev0). This arises due to numba requiring an older version of numpy than the one nionswift uses.
As indicated by the hyperspy community, in the present case nionswift seems not to follow the depreciation policy of numpy (https://numpy.org/neps/nep-0029-deprecation_policy.html).
The actual solution is to fork hyperspy and remove the calls to numba, which is not very handy. Also, I guess numba could be of some interest for nionswift application field.

cmeyer, 2021-11-28:
I didn't test this recommendation - but I think you can override the nionswift dependency to use the older numpy without any issue. The reason for the bump to numpy 1.21 is for numpy compatibility with Python 3.8 protocols; and that is only required for nionswift-instrumentation-kit and to pass Python typing tests. But, again, I haven't tested this directly. I'll look at it more closely tomorrow and see if we can back off the numpy requirement for nionswift only - nionswift-instrumentation-kit will continue to require the new numpy, unfortunately.

cmeyer, 2021-11-28:
Also: 
- https://github.com/numba/numba/pull/7483
- https://github.com/numba/numba/issues/7562

kociak, 2021-11-29:
Thanks. #769 is an excellent idea that broadens the restricted scope of the present issue
As for the current issue, my workout is to fork hyperpsy and remove any dependence to numba. A short term workaround only of course.

> I think you can override the nionswift dependency to use the older numpy without any issue

sounds like a good short term workaround (until passing to the on-microscope nionswift version, that encompasses the instrumentation-kit). However, I am unsure _how_ to override dependencies (any link welcome!)
Thanks for the link to the numba PR. Note sure when this version will be used in hyperspy

cmeyer, 2022-01-16:
Closing this with the [release of numba 0.55](https://numba.discourse.group/t/numba-0-55-0-and-llvmlite-0-38-0-final-release/1136).

closed 2022-01-16 by cmeyer