nion-software/nionswift #769: Establish broad set of libraries for which nionswift should be co-installable

cmeyer, 2021-11-28:
Some obvious libraries would include scikit-learn, scikit-image (a nionswift pacakge for this is already in progress), hyperspy, numba, pandas, xarray, matplotlib, seaborn, NLTK, tensorflow, keras, PyTorch, Eli5, Theano, etc.

This might be as simple as running the tests in an environment with all of this installed. It might also include specific tests for uses those libraries within Nion Swift.

See #768 too.

TomaSusi, 2021-11-29:
We'd also very much appreciate compatibility with abTEM (https://github.com/jacobjma/abTEM) since we – and hopefully more and more of the Python EM community — often run both software, and it would be fairly easy to implement abTEM-based plugins into Swift. But by making sure that the underlying libraries used by us and projects like Hyperspy work should go a very long way in ensuring this.

TomaSusi, 2021-11-29:
Py4DSTEM would also be great but those guys should also rethink their exact version requirements for some packages.

cophus, 2021-12-02:
@TomaSusi as of today the relaxed dependencies for py4DSTEM should be pushed to main / pypi - hopefully this will make things a lot easier going forward!