nion-software/nionswift #773: Minimal dataitem selector example would be useful

kociak, 2021-12-03:
In many cases, when writting scripts or plug-ins (or computations), it is necessary for the user to choose one to several dataitems to perform operation with/on them. Some computation interfaces are nice (where you can e.g drag and drop dataitems on panels/windows). I could not find  a minimal example of such a "dataitem selector". This would serve the purpose of the "get2images" or alike function of digital micrograph, but I guess the drag and drop/ visual version of it is quite useful.

cmeyer, 2021-12-04:
The approach we've been taking so far is to select two images by clicking on one display panel and then shift-clicking on a second display panel. I'd also like to have exactly what you suggest - but somewhat extended so it was actually a "wizard" that allows you to configure a computation before actually performing it. The "wizard" could have one or more pages and would be a modeless, floating window, so you could do other things while configuring it.

Either way, this is a great request and I'll write up something about it soon. Thanks.

kociak, 2021-12-05:
The idea of a wizard sounds like honey to my ears.
Using a double-selection is not appealing for me in a script, because simply I don't know how to pause a script, letting it waiting for inputs or validation.
In a plug in, i can do it easily with my own buttons. However, what I don't like is that I have trouble remembering which one I selected first - probably not an issue for everybody. So I am planning to make two "select" button (one per dataitem for a 2 parameter function).
I just realized the selected items are numbered if they appear in a display panel... lovely! This would really help me in the future. Having the number also on the miniatures would be fantastic as well!

kociak, 2021-12-13:
Just experimenting with the idea of having several selected items. Using the `api_broker` does not allow to make it in the script environment. We need to use the `Facade` instead:
```
api = api_broker.get_api(version="1.0", ui_version="1.0")
#the following line causes an exception
print(api.application.document_controllers[0]._document_controller.selected_data_items)
```
does not work while the following works

```
api=Facade.get_api("~1.0", "~1.0")
print(api.application.document_controllers[0]._document_controller.selected_data_items)
```

works

cmeyer, 2022-05-21:
- https://github.com/nion-software/nionui/issues/54