nion-software/nionswift #777: Creating a docker container of python nionswift API

aneesalnajjar, 2021-12-16:
Hi @cmeyer,
Can you provide a docker container that containerizes all the neionswift applications, including the sim and use cases?

Anees Al Najjar

aneesalnajjar, 2021-12-16:
> Can you describe your use case? There are typically too many variants (3 platforms x 3 versions of Python) for us to usefully provide a docker container.
>
> We already provide an installer on macOS and Windows which bundles everything. The most recent downloads are:
>
> nion.com/swift/files/NionSwift-0.16.1.dmg
> nion.com/swift/files/NionSwift-0.16.1.msi
>
> (note: you have to copy and paste the links into your browser because GitHub does not allow direct downloads from links within issues).
>
> The builds for the installer are here: https://github.com/nion-software/nionswift-installer
>
> In addition, we have the .devcontainer Dockerfile<https://github.com/nion-software/nionswift/tree/master/.devcontainer> that creates a codespace<https://github.com/features/codespaces> that I explored for use with documentation; you could use that as a starting point.

Morning Chris,

So the idea is to have scientific data and nionswift simulator in different places, and the best way to implement that is via having containers for the data and the nionswift and sim as well. For the latter just asking on a python3 API platform. Don’t have a certain use case, but if you have some implemented ones, appreciate to the include them.

Thanks for your time and effort.

Anees

cmeyer, 2021-12-22:
In order for this to be useful, the following issues should be addressed:

- [ ] #429 
- [ ] #539

simon-77, 2022-09-06:
Dear all

I just created a docker container for running Nion Swift GUI in an encapsulated way.

If you are interested, check out my github repository:
https://github.com/simon-77/nionswift-docker

Best regards,
Simon