nion-software/nionswift #782: Line profile on multiples images

yiranlus, 2021-12-21:
Is it possible to use the same line profile on different images so that intensity on different images at the same positions can be compared? It would be good if all line regions kept their positions updated with each other when I am exploring the data.

cmeyer, 2021-12-21:
Thanks for this request. Unfortunately this is not possible right now - but we've also been wanting this for a while, too.

Some related items that need to be done:
- [ ] #548 
- [ ] #113 
- [ ] #178

cmeyer, 2024-02-27:
BPS Comment from 2024-02-06:

Copying line profile parameters from one image to another.

This may sound like a small or specialized thing. but it comes up all the time. There is a technical workaround to make the same line profile on two images, but it is very tedious, and it becomes practically unworkable as the number of data items you want to compare goes up.

I really want to emphasize this one. Having a quick workflow for this would be huge.

It could be as simple as being able to copy all the line profile parameters at once from the inspector, and paste them all at once into the inspector for another line profile.

I think the ideal workflow would be that if you select and ctrl V a line profile graphic, and paste it onto another data item, it should automatically create the same line profile on the new data item if the dimensions are compatible.

kociak, 2024-02-27:
I am not sure it is the right issue for that but having mirrored elements, such as ROI ( especially pick tools for SI) is in general something needed. The practical case for SI is browsing an HAADF with a pick tool and having the corresponding spectrum from the corresponding SI updated. Same between ie two or more SI, those SI being for example raw and data analyses versions of the raw one. In our lab we keep talking about this feature again and again ...

cmeyer, 2024-02-27:
Moving main discussion of mirroring graphics to #548

cmeyer, 2024-02-27:
Closing as duplicate of #548

closed 2024-02-27 by cmeyer