nion-software/nionswift #783: Computation fails to call execute and commit when inputs are scalar

kociak, 2021-12-27:
It seems impossible for me to create a viable computation with scalar values (at least numerics). Using `data_items` and `graphics` as `inputs` does work (at least when there are only 2), but not using `real`. In the following code (you need to call `register_OffsetAtZero_process` and maybe install it in a menu) only the version with `data_items` and `graphics` works. Any other combinaison with other `inputs` simply call the `__init__` `execute` `commit`sequence once and then stops. No window pops up on `cmd+E`.

```
# imports
import numpy as np

# local libraries
from nion.swift import Facade

class OffsetAtZero:
    

    def __init__(self, computation, **kwargs):
        """Initialize the computation."""
        self.computation = computation
        self.api=Facade.get_api("1","1")
        print("__init__")
    def execute(self, src,average_interval):

        print("execute")
        return
        
    def commit(self):

        print("commit")
        return
        
def register_OffsetAtZero_process(api):
    api.register_computation_type("Compute.OffsetAtZero", OffsetAtZero)

def attach_OffsetAtZero(api, window):
    target_data_item = window.target_data_item
    interval=target_data_item.add_interval_region(0.1,0.2)
    offset_data_item = Facade.get_api("1","1").library.create_data_item_from_data(target_data_item.xdata.data, title="{} offset".format(target_data_item.title))
    
    # the following line does not work: __init__, excecute and commit are just call once
    computation=api.library.create_computation("Compute.OffsetAtZero", inputs={"src": 10,"average_interval":10}, outputs={"offset": offset_data_item})
    # the following line does  work: 
    computation=api.library.create_computation("Compute.OffsetAtZero", inputs={"src": target_data_item,"average_interval":interval}, outputs={"offset": offset_data_item})
    #anytime another "input" is added, the computation does not work
        
```

cmeyer, 2022-01-18:
The computations represent connections between items (data items, graphics, displays, etc.) within a project. In this case, the non-working computation is not connecting any items together, i.e. it has no source items, but only scalar values. So the computation runs once; but after that there are no input items on which to watch for changes, so it is never called again automatically. The computation edit window does not appear because the computation isn't attached to an item , data item or otherwise.

Technically, you could use `computation.set_input_value("src", 12)` to trigger the recompute, however there would still be no way to edit the computation.

My suggestion, at least for now, is to always pass the data item as "src", even if it isn't used. That way the computation is always making a connection between that data item and something else. That will also make 'edit computation' work. If you want to have an overriding scalar value, pass it is as "src_override_value" or something similar.

Also, internally we also have items called 'data structures' for this purpose - items in which to store scalar values. However, they are deprecated and will not be supported in the future; so I don't recommend using them. My plan is to merge them [data structures] with computations (at least the storage part) and data items so that we have a standard way of handling this situation. This is a larger topic, so I'll leave it at that.

As I mentioned before, the computations and API for the computations are considered to be internal objects and you should use them with caution because they will be changing to a more permanent form in the next few months.

kociak, 2022-01-18:
sorry, there was an obvious typo, my very bad.
you should have read

```
# imports
import numpy as np

# local libraries
from nion.swift import Facade

class OffsetAtZero:
    

    def __init__(self, computation, **kwargs):
        """Initialize the computation."""
        self.computation = computation
        self.api=Facade.get_api("1","1")
        print("__init__")
    def execute(self, src,average_interval):

        print("execute")
        return
        
    def commit(self):

        print("commit")
        return
        
def register_OffsetAtZero_process(api):
    api.register_computation_type("Compute.OffsetAtZero", OffsetAtZero)

def attach_OffsetAtZero(api, window):
    target_data_item = window.target_data_item
    interval=target_data_item.add_interval_region(0.1,0.2)
    offset_data_item = Facade.get_api("1","1").library.create_data_item_from_data(target_data_item.xdata.data, title="{} offset".format(target_data_item.title))
    
    # the following line does not work: __init__, excecute and commit are just call once
    computation=api.library.create_computation("Compute.OffsetAtZero", inputs={"src": target_data_item,"average_interval":interval,"dum":10}, outputs={"offset": offset_data_item})
    # the following line does  work: 
    computation=api.library.create_computation("Compute.OffsetAtZero", inputs={"src": target_data_item,"average_interval":interval}, outputs={"offset": offset_data_item})
    #anytime another "input" is added, the computation does not work
        
```
I know you won't maintain this version of computations, but I just need in my application to be able to set an integer (this is for the implementation of deconvolution, which is now missing, and I hope I can release it after testing by the crew here to the community). And I can't, which really puzzles me

kociak, 2022-02-17:
Thanks
this works...

closed 2022-02-17 by kociak