nion-software/nionswift #801: Apple M1 Support

tcpekin, 2022-03-03:
I am getting an error when installing Nion Swift as not all the submodules (not sure if that is the right word) are M1/Arm compatible. Nion Swift can be installed and is available on the following platforms: 

[image]

But nionswift-tool cannot be installed as it is only specified for x64 Macs. 
[image]

I do not know if Rosetta the transpiler can handle things like installed packages from conda, so I don't know if this is as simple as changing a flag to the conda upload as to which processors are supported, or it would require more work on your end, but it would be nice to be able to install Nion Swift on newer Macs.

Thanks!

cmeyer, 2022-03-06:
TL;DR: No M1 support yet and its a few months away, depending on conda-forge community support. And you're using a less performant UI when using pyqt. Read on below...

Nion Swift defines a user interface API that is required to be able to run its UI. The UI API is implemented in almost identical ways by `nionswift-tool` or by `pyqt` along with a related Python file `PyQtProxy.py`.

The two implementations are very similar, one being a C++ implementation and one being a Python implementation using the underlying `pyqt` (or `PySide2`) library. The difference is in display optimization - the C++ implementation is capable of significantly faster raster display updates, which is particularly important for live acquisition.

Also, Qt has stopped releasing new version of Qt 5.15. Unfortunately, they made this choice right around the time the M1 processor started being used; so with Qt 5, we're stuck using Qt 5.15.3 and that does not support the M1. There are community efforts to support the M1 on Qt 5.15, but so far, nothing is available.

Qt also has made available Qt 6 with support for the M1 processor. However, up until now, Qt 6 has not been available for many of the community distributions that we use (i.e. conda-forge and conda), so I've held off making a new UI API implementation for Qt 6 until those are available. Again, they're slowly progressing and I'm hopeful everything will be available soon.

Another potential, but expensive, solution in the future is to make purely native (macOS, Windows, Linux) implementations of the UI API and take Qt out of the equation. This would be time consuming, so I'd probably only advocate for this as a last resort if Qt continues to narrow their licensing terms for open source.

As a fun thought experiment, too, I've considered making a web browser canvas-based UI API implementation and then be able to run Swift (and any other `nionui` based projects) in the browser. It's a dream, but it would be quite time consuming and performance wouldn't be great, but kind of fun to think about!

Also, for reference, the UI API interface is [defined here in `UserInterface.py`](https://github.com/nion-software/nionui/blob/6c081d9cc1b263a3305c852cfb7ed5c623b676f8/nion/ui/UserInterface.py#L3658). It's somewhat more complex than that single class, but if you're interested, it gives you an idea of how the UI API is defined.

tcpekin, 2022-03-07:
Hi Chris, thanks for the info, it's always interesting to see behind the scenes. Right now, Swift works on my M1 Mac using the steps described above, so I wonder if it is useful putting this thread behind a link in the documentation, for anyone else who has the same issue as I do. Would that make sense?

tcpekin, 2022-06-29:
Hi Chris, any updates on this?

tcpekin, 2022-06-29:
Great to hear! Thanks a ton Chris I'll give it a shot.

tcpekin, 2022-09-09:
Hi Chris, I tried a bit and pretty much failed to get Swift running on my M1 Mac... both with installing from github and installing the nionui tool. Is this just operator error?

cmeyer, 2022-09-09:
I'm (hopefully) releasing this next week ... but if you want you can use `nionui-tool` (note: this is different from `nionswift-tool` in that you have to pass the application name to launch.

You can download the proper tool (macOS, probably Python 3.10) here:
https://github.com/nion-software/nionui-tool/actions/runs/3012405255

You will need to have an arm64 conda installation with `nionswift` installed. Something along the lines of:
`CONDA_SUBDIR=osx-arm64 conda create -n my-swift nionswift nionswift-io`

Then launch it using (the app name and the paths for your conda installation will be different, of course):
`Nion UI Tool.app/Contents/MacOS/NionUILauncher /path/to/your/conda/env/my-swift nionui_app.nionswift`

tcpekin, 2022-10-24:
Bumping this again to say that these instructions worked for me, it was a little confusing but so far everything is running great!

cmeyer, 2022-10-24:
The standard nionswift-tool on the nion channel should work now too:

`CONDA_SUBDIR=osx-arm64  conda install -c nion nionswift nionswift-tool`

I'm waiting to close this until I can build an Apple M1 prepackaged installer version; and also the conda-forge arm64 too. Both of those require cross compiling, which I almost have working, but not quite!

cmeyer, 2022-12-21:
Qt macdeployqt issue (wrt conda-forge): https://bugreports.qt.io/browse/QTBUG-109572