nion-software/nionswift #807: Accessing multidimensional Swift data in external Python analysis programs

TomaSusi, 2022-03-18:
I can't imagine we in Vienna are the only ones who are encountering the need to analyze large multidimensional datasets acquired in Swift in an external Python analysis program – for example feeding a time series of images to a machine-learning analyzer, or a 4D-STEM dataset e.g. into py4DSTEM for ptychograpic reconstructions.

Currently, the only supported way to do this is to export a file and import that, but since some realistic datasets can reach tens or even hundreds of GB in size, it seems silly to force the user to create a duplicate file.

We've hacked together a plugin that copies the full path(s) to one or multiple data items selected in the Swift GUI, which can then be directly read into analysis via numpy.load(). The basic code is rather simple, and Jacob Madsen also made a plugin which is basically a button to run the code:
https://github.com/jacobjma/nionswift-path-exporter

However, this seems something that should be supported in Swift on a lower level, perhaps via a menu item or the right-click contextual menu.

cmeyer, 2022-03-18:
Here are two related issues that have already been filed:

- #88 
- #539 

My suggestion is that #539 would be the best approach.

The problem with accessing the data files within a Swift project directly is that the internal format is Swift projects is not something we can guarantee will be stable. We expect to be able to change the internal formats without regard to 3rd party software.

What we _can_ do is provide a Python package (maintained by Nion) to open a Swift project (without requiring the UI to be launched) and then access individual data items within the Swift library.

First, does that sound reasonable? If not, do you have an alternate suggestion?

Second, if that sounds reasonable, what sort of API would be best to provide for large data items? Sometimes those items will be stored within h5py files, sometimes within zip files, and sometimes within npy files. Would it work to provide a Numpy-like array? Does it need to be writeable or only read-only?

If this approach (#539) sounds reasonable, we can move this conversation/responses to that issue instead.

TomaSusi, 2022-03-18:
Thanks for linking to existing issues!

In principle an API sounds like an elegant solution, but I'm a little uncertain if I understand how the user can select the right data item without a GUI? Can't data items have identical names (the titles in the GUI)?

I personally think that a numpy-style array would be ideal regardless of the underlying file format, though I am not sure if this scales to really large datasets?

Read-only would IMO definitely be desired for these kinds of uses, if the user wants a mutable file they should just export.

cmeyer, 2022-03-18:
What I mean by "no UI" is that the package allowing you to access the data would not require a GUI to be launched. In more detail: right now, you can launch Swift and easily get access to the data of any data item via the existing API. But if you want to write a script that functions from the command line and has no GUI, it is difficult to do that right now. The package proposed in #539 would allow that script to be easy to write. The data item would have to be specified by a UUID.

Maybe you can explain more of what you're trying to accomplish... are you trying to do something while running the Swift UI? or something separate/offline without the UI?

TomaSusi, 2022-03-18:
Specifying the data item by the UUID that the user never normally sees is a potential pain point here, plus the fact that there's some amount of code that is necessary to recover the file that is not obvious to the user.

What we are trying to accomplish is to take a saved Swift library, pick data items that we want to analyze further in an external program, and have an easy way to import the underlying data into that external code. So there are two issues:

1) how the user select the data items
2) how the external code accesses the files

For 1), the GUI is useful, though an API-based approach would also acceptable (I know there is a menu item to reveal the UUID). For 2), ideally there should be no need to launch Swift or have any sort of GUI interaction, just some np.load() type command that allows the analyses to be re-run and modified in e.g. a Jupyter notebook.

cmeyer, 2022-03-18:
With #539 you could iterate through the data items, present them to the user in your own UI (Jupyter notebook or whatever), and then get NumPy array-like data arrays for each desired data item. The Swift UI would not need to be running. I'm assuming that I could provide the `data_item_to_svg` method in the external API - the biggest difficulty there is probably measuring text sizes without a GUI, but it should be possible.

At the low level, the data items can always be looked up by UUID. And the UUID provides a way to persistently store a reference to a data item within a particular project. I don't envision using the UUID as a typical end user thing; although I have considered adding a URL type to reference data items in a project so that it is possible to email comments like "please look at this (item url) in this project" and the user can use the URL to open the item within a particular project.

TomaSusi, 2022-03-20:
Thanks, I do think having some easily communicable way to share links to data items would be welcome.

What you propose in terms of an external UI to iterate through data items does sound like a decent solution, though I would then expect that the API provides a simple function to do this.

cmeyer, 2022-03-20:
It would be something similar to what is available now:

```
for data_item in project.data_items:
  if is_the_data_item_we_want(data_item):
    do_something_with_data(data_item.data)
```

TomaSusi, 2022-03-21:
Yes but how does one actually do the "is_the_data_item_we_want" step in practice?

tcpekin, 2022-03-21:
TL;DR I think #539 sounds great.

While I haven't used the package in the @TomaSusi 's original comment, Alberto in our group has developed something similar, but maybe a bit more low level. My workflow using Alberto's code is to copy the UUID for every dataset I'm interested in with the GUI, save those in a text file/python list and then iterate over those using Alberto's loading function, which essentially does a UUID lookup of all the data items and then loads the one that matches, as a numpy file. I think that is what Chris is proposing, officially supporting some sort of API/package for the lookup, when given a UUID or specific key value pair.

I think the key part of `is_the_data_item_we_want` is that that is a user defined function - it looks at data_item's UUID or metadata to see if it is actually interesting, and then executes. 

Pseudo-python I imagine it being something like this:

```python
import numpy as np
from nion_api_toolbox import api

f_project = 'path_to_project`
UUID = 'UUID_you_want_to_match' # or other key value pair that matches the data you want to look at

project = api.open(f_project)

for data_item in project.data_items:
  if data_item.metadata['UUID'] == UUID: # this could be any of the metadata though
    do_something_with_data(data_item.data) # or just loaded_data = data_item.data
```

This essentially matches what Alberto's code can already do, and I find it very handy. His code can also sort the data a variety of ways, so you can load data created between certain times, etc. But I mainly use it to look up data using UUIDs.

I would prefer if the `api.open` command defaulted to read only, as I use the Nion Swift project as my ground truth and don't like to modify the data it contains. Maybe there could be a way in the api to import data that had then been generated using the `loaded_data`. Something like `project.import(numpy_array, metadata_dict)`, that would add the numpy array and metadata to the Nion Swift project, but that would go well beyond what I personally need, and I think could cause a lot lot lot of bugs or edge cases to deal with.

TomaSusi, 2022-03-30:
I just stumbled upon the Hyperspy Swift Library, which seems to do pretty much exactly what you prosed:
https://github.com/hyperspy/hyperspy_swift_library

francisco-dlp, 2022-10-25:
In case that somebody is still interested on this, we have just fixed [HyperSpy Swift Library[(https://github.com/hyperspy/hyperspy_swift_library) to work with the latest version of Nion Swift.