nion-software/nionswift #810: Strange behavior, when using nionswift through and ssh tunnel with X11 forwarding

w-markus, 2022-03-28:
Dear all,

my current setup:

two computers, A and B, respectively, both running debian's bullseye distribution with up-to-date packages.

Current (Mini)conda, created an environment on "B" running python 3.8.

On "PC A" MS "vscode insider" that connects to "PC B" using vscode's remote plugin, essentially using ssh. My ssh configuration has X11Forward set to Yes.

On "B" there is a local copy of a yet private gitlab repository which I use for developing a plugin.
"nionswift" and some other (dependent) modules are inside this repository as submodules (via ```git submodule add <repo url>```) and all installed by

```bash
pip install --no-deps -e .
```
inside my conda environment.

Now, upon starting nionswift (inside vscode [using the "qt, debug" configuration from  ```launch.json``` from this nionswift repo here], or from the command line), logging starts and finally the window pops up, if and only if ```"$HOME/.local/share/Nion/Nion Swift/Profile.nsprojs"``` is not present, i.e. has been deleted by me beforehand.

Debugging shows, that the script execution stops in
```PyQtProxy.py``` in line 2952:
```python
document_window.show()
```
and this without any further notice (no line from some logger script, nor an exception)!

```document_window``` at that time is a
```<nion.ui.PyQtProxy.PyDocumentWindow(0x5584f70da730) at 0x7f9fc0711e40>```

So I am puzzled. As a workaround I use a ```tasks.json``` file that deletes the Profile file before lauching nionswift, but this not so nice.

If I can provide more information, please let me know.

Thank you very much and best wishes
Markus

cmeyer, 2022-04-04:
I haven't spent much time with VSCode myself - `launch.json` was a quick attempt to try it out.

What are trying to accomplish? Remote debugging? Or something else?

w-markus, 2022-04-05:
Thanks Chris!

Our direct electron detector is attached to a server, which we access remotely only. So yes, it's remote debugging, we want to do. I think ```launch.json``` works fine.

But I am quite sure, that this is more related to the X-Forwarding part, as the problem occurs also, when starting nionswift from a command console (without vscode whatsoever).

Brow71189, 2022-04-05:
Just to add some details to this, I have tried to reproduce this problem but wasn't able to. What I did is install Nion Swift into a virtual machine running Ubuntu and connect to that one via ssh with X-forwarding. With this setup, everything worked fine for me.

So I guess there must be some specifics in your setup that trigger this problem. Once difference is that I installed the "release" version with conda instead of a developer installation. The other might be the specific OS, I used Ubuntu and you are using Debian.

I'll try this again with a configuration closer to yours and see if I can reproduce it.

A couple of questions for this:

1. You are using a developer installation of Nion Swift, right? This means you cloned the repositories and are running from source, right?
2. Are you using conda for the python installation on the remote server? If so, could you post the output of `conda list` run when your environment that you use for running Swift is activated?
3. Are you using `pyqt` or `nionswift-tool` to launch Swift?
4. Have you tried installing the release version of Swift on your remote machine following the instructions here: https://nionswift.readthedocs.io/en/stable/installation.html#linux Could you give that a try and see if it causes the same problem?

Thanks!

w-markus, 2022-04-05:
with respect to the questions of @Brow71189:

1. Yes cloned as submodules into to "my" repository used to develop a detector plugin:

```bash
echo -e "nionswift \n nionswift-io \n nionui \n niondata \n nionutils \n nionswift-instrumentation-kit" | while read REPO ; do  echo ${REPO} ; git submodule add https://github.com/nion-software/${REPO}.git ; cd ${REPO} ; pip install --no-deps -e . ; cd .. ; done
```
2. conda list:
```bash
(nion-dev) cri@CZ205305KQ:~/Software/nionswift-software$ conda list
# packages in environment at /home/cri/Software/conda/miniconda/miniconda3/envs/nion-dev:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
alsa-lib                  1.2.3                h516909a_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.zoneinfo        0.2.1            py38h497a2fe_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2021.10.8            ha878542_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
cython                    0.29.28          py38h709712a_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
expat                     2.4.7                h27087fc_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.96              h8e229c2_2    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freetype                  2.10.4               h0708190_1    conda-forge
gettext                   0.19.8.1          h73d1719_1008    conda-forge
giflib                    5.2.1                h36c2ea0_2    conda-forge
gst-plugins-base          1.20.1               hcf0ee16_1    conda-forge
gstreamer                 1.20.1               hd4edc92_1    conda-forge
h5py                      3.6.0           nompi_py38hfbb2109_100    conda-forge
hdf5                      1.12.1          nompi_h2386368_104    conda-forge
icu                       69.1                 h9c3ff4c_0    conda-forge
imageio                   2.16.1             pyhcf75d05_0    conda-forge
jbig                      2.1               h7f98852_2003    conda-forge
jpeg                      9e                   h7f98852_0    conda-forge
keyutils                  1.6.1                h166bdaf_0    conda-forge
krb5                      1.19.3               h3790be6_0    conda-forge
lcms2                     2.12                 hddcbb42_0    conda-forge
ld_impl_linux-64          2.36.1               hea4e1c9_2    conda-forge
lerc                      3.0                  h9c3ff4c_0    conda-forge
libblas                   3.9.0           13_linux64_openblas    conda-forge
libcblas                  3.9.0           13_linux64_openblas    conda-forge
libclang                  13.0.1          default_hc23dcda_0    conda-forge
libcurl                   7.82.0               h7bff187_0    conda-forge
libdeflate                1.10                 h7f98852_0    conda-forge
libedit                   3.1.20191231         he28a2e2_2    conda-forge
libev                     4.33                 h516909a_1    conda-forge
libevent                  2.1.10               h9b69904_4    conda-forge
libffi                    3.4.2                h7f98852_5    conda-forge
libgcc-ng                 11.2.0              h1d223b6_14    conda-forge
libgfortran-ng            11.2.0              h69a702a_14    conda-forge
libgfortran5              11.2.0              h5c6108e_14    conda-forge
libglib                   2.70.2               h174f98d_4    conda-forge
libgomp                   11.2.0              h1d223b6_14    conda-forge
libiconv                  1.16                 h516909a_0    conda-forge
liblapack                 3.9.0           13_linux64_openblas    conda-forge
libllvm13                 13.0.1               hf817b99_2    conda-forge
libnghttp2                1.47.0               h727a467_0    conda-forge
libnsl                    2.0.0                h7f98852_0    conda-forge
libogg                    1.3.4                h7f98852_1    conda-forge
libopenblas               0.3.18          pthreads_h8fe5266_0    conda-forge
[... 42 lines cut ...]
pyxsp                     3.0.1                     dev_0    <develop>
qt                        5.12.9               h1304e3e_6    conda-forge
readline                  8.1                  h46c0cb4_0    conda-forge
scipy                     1.8.0            py38h56a6a73_1    conda-forge
setuptools                60.10.0          py38h578d9bd_0    conda-forge
sqlite                    3.37.1               h4ff8645_0    conda-forge
tk                        8.6.12               h27826a3_0    conda-forge
tzdata                    2022a                h191b570_0    conda-forge
[... cut ...]
cd /home/cri/Software/nionswift-software ; /usr/bin/env /home/cri/Software/conda/miniconda/miniconda3/envs/nion-dev/bin/python /home/cri/.vscode-server-insiders/extensions/ms-python.python-2022.4.0/pythonFiles/lib/python/debugpy/launcher 45803 -- -m nionswift --ui qt --no-fallback True 
```
I think I came across ```pyqt```/```nionswift-tool``` in the docs, but could you kindly point me towards this again, please?

4. What I tried was installing ```nionswift```, ```niondata```, etc., by use of ```conda install```. However, I remove it again, since I could not debug my code in vscode taking avantage of e.g. on-the-fly documentation when hoovering over class names. But I can give it a try.

cmeyer, 2024-02-26:
Closing this for inactivity. If this is still an issue that doesn't have a workaround, please re-file.

closed 2024-02-26 by cmeyer