nion-software/nionswift #814: "Crop" computation gives inconsistent results for specific data/crop shape combinations

Brow71189, 2022-04-03:
One example:

Create some data with shape ``(2672, 4008)`` (h, w)
Run ``Processing -> Transform -> Crop`` computation on it
Resize the crop graphic to ``(512, 512)`` pixels size

The result of this crop computation has now a shape of ``(512, 511)`` pixels whereas it should of course be ``(512, 512)`` as requested by the user setting up the computation with target shape ``(512, 512)``.

I believe this happens due to rounding errors because we use fractional coordinates for graphics internally. But this should of course work because sometimes you need to crop your data to a specific shape and being one pixel off can break further processing steps.

cmeyer, 2022-04-03:
This would be fixed by #109. I'll leave this open, but #109 will be the main issue.

Also, I think the specific shape will depend on the origin of the crop rectangle. Can you supply the specific origin and size in fractional coordinates?

Brow71189, 2022-04-07:
The center was left as it comes out when you run "Crop", so the graphic is centered on the source image. Below are the raw numbers for the graphic after changing its size to (512, 512) pixels.

```
>>> graphic.center
<nion.utils.Geometry.FloatPoint object at 0x7fe780bac640> (x=0.49999999999999994, y=0.5)

>>> graphic.size
<nion.utils.Geometry.FloatSize object at 0x7fe780b595e0> (w=0.1277445109780439, h=0.19161676646706588)

```

ndellby, 2022-04-07:
So you had a base area of 4008 by 2672. That is a weird choice but 
numbers spot on, is that a screenshot off a 4K monitor?

Brow71189, 2022-04-07:
No that was just some image from the internet that I was processing and happened to trigger this problem.

cmeyer, 2025-07-24:
Bumped priority.

cmeyer, 2025-07-24:
This comment is referring to multidimensional crop, but the issues are probably similar. Adding this comment here from @bentps

> I'm using multidimensional processing to crop a 4D dataset along the detector directions. I haven't checked if you get the same behavior cropping a 2D image.
> 
> I want to crop so that I have exactly 100x100 pixels on the detector axes in the cropped data. I put a square graphic on the 4D data, set the size in pixels in the inspector to be 100x100 pixels. I set the position of the graphic to be an integer number of pixels (which shouldn't be necessary, but mentioning that I'm doing that for now). The resulting cropped data item is 99x100 pixels on the cropped dimension.
> 
> I then set the graphic to be 100x100.1 pixels, which results in a 101x100 pixel cropped data.
> 
> Then I set the graphic back to 100x100, which results in a 100x100 pixel cropped data.
> 
> This is related to the discussion on github about the number of pixels selected by a mask. The common theme here is that the number of pixels selected needs to be separated out from the question of which pixels. If I use a 100x100 pixel graphic, it should result in a 100x100 pixel crop region however the graphic is placed on the detector. The behavior I'm seeing with 99x100 crop region from a 100x100 graphic shouldn't be possible, in addition to the fact that it should not have the weird hysteresis behavior that I've described.

Ion-e, 2025-07-28:
Also see:

https://github.com/nion-software/nionswift/issues/1626
https://github.com/nion-software/nionswift/issues/1627
https://github.com/nion-software/nionswift/issues/1628