nion-software/nionswift #846: Display resolution may be incorrect on secondary monitors

cmeyer, 2022-06-01:
This needs to be confirmed. The problem will show up most recognizably as either too high or too low resolution text in the display panels. Also if a window spans multiple displays, it may appear strange.

Note: Qt has no mechanism to notify specific widgets (PyCanvasWidget) when screen resolution changes.

icbicket, 2022-06-03:
I've noticed that if I use a 2x User Interface scaling on my secondary monitor (offline), Swift will have enormous font in the display panel titles. If I change the DPI of the monitor instead, it looks reasonable to me. I'm on Linux Mint 20.2 with Swift 0.16.2.

cmeyer, 2024-02-23:
Probably fixed. Reopen if seen again.

closed 2024-02-23 by cmeyer