nion-software/nionswift #932: export dialog behaves differently for a single window vs. multiple

tylerharvey, 2023-01-18:
This dialog window is only available for multiple-window export.
[image]
I can see design reasons for this, but for consistency of use it would be really nice to also have the same export options for single-window export. At the moment, for consistent naming, a user must make sure to always export multiple windows.

bentps, 2023-04-10:
I want to second this request. In addition to the dialog window options being available, the multi-data-item export puts dashes between the words of the data item title, while exporting a single data item uses spaces.

jamesrussell216, 2024-06-14:
@KRLango is going to start working on this one now. Just defining a few bits for prosperity...

- In moving the single item export code over to work off the multi item export, as @bentps says that will lead to single items exporting with the multi item export naming convention. Which as far as I can see is probably a good thing but I thought I'd explicitly state it unless anyone can see a problem!
- @KRLango is going to "tidy" the UI up a little as well.

jamesrussell216, 2024-06-17:
Fixed by #1070

closed 2024-06-17 by jamesrussell216

Ion-e, 2024-07-02:
Blocked by https://github.com/nion-software/nion-internal/issues/136

jamesrussell216, 2024-07-03:
So I'd argue @Ion-e that #136 is an extension on the original spec for this - a worth while one, but an extension none the less, so I'd further argue if the feature is otherwise working as intended this is completed and we'll add #136 to the to do pile?

cmeyer, 2024-07-03:
This issue is closed. Can you open a discussion for further discussion of unrelated items?

jamesrussell216, 2024-07-03:
It is not an unrelated item and it's a simple question - I don't think it needs a discussion item.

Ion-e, 2024-07-04:
Verified as implemented, thank you!