nion-software/nionswift #939: Hitting return on a line plot data item scales x-axis incorrectly

Brow71189, 2023-02-07:
If you double click on the x-axis, the scaling is correct:

[image]

Hitting return adds some extra margin:

[image]

This was reported by a customer but if this is intended behavior feel free to just close this.

TomaSusi, 2023-02-07:
Customer here ;) This certainly does not feel like correct behavior for a user.

cmeyer, 2023-02-07:
If intervals are selected, the return key will zoom to those selected intervals and is supposed to add padding on either side of the interval so that the intervals don't end up at the edges of the display. When no intervals are selected it probably makes more sense to have the behavior the same as double clicking on the axis.

TomaSusi, 2023-02-07:
Right, makes sense with the intervals but that's not the case here.

cmeyer, 2023-02-07:
Also this code (see 0556bbe9cf873c2b7180b1da8df9025ec212ddad) deals with multiple layers, so the fix is relatively easy, but not trivial.

commit 81b2a5bd (2023-06-22): Fix #939. View to interval on line plot handles special case of no-interval same as double clicking axis.
closed 2023-06-22 by cmeyer