nion-software/nionswift #940: Workspace and Window layout should be saved more often than just on exit.

Brow71189, 2023-03-06:
For example, if you had Swift maximized when you last closed it, this will be the saved state. If you now change to windowed mode while it is up, and then minimize and restore Swift it will return to maximized instead of windowed.
The same happens for the panel arrangement within Swift: After minimizing and restoring the Swift main window you get the arrangement from when you last exited Swift, not from when you minmimized it.

jamesrussell216, 2024-07-16:
That just sounds like a bug to me.

cmeyer, 2024-07-16:
I changed the title too.

Ion-e, 2024-08-08:
Duplicate behaviour from https://github.com/nion-software/nion-internal/issues/109#issue-2357094971

cmeyer, 2024-08-08:
Why was this closed as completed? It is still an open issue.

Ion-e, 2024-08-12:
Issue partially fixed, however there seems to still be a corner case where this issue happens:

Screen 1 >start with swift maximised.
Drag Swift to screen 2 and drop at the top corner of the screen (This should move swift to screen 2 and maximise it in screen 2).
Minimise Swift.
Reopen Swift.
Swift will reopen in window 1.

jamesrussell216, 2024-08-14:
@Brow71189 are you happy with closing this one in favour of us making a new, much smaller ticket for the corner case where this still doesn't work (which is hopefully much much less of a problem than this was!)?

Brow71189, 2024-08-14:
If the initial bug is fixed, then yes, I'm fine with closing this.
Is there a commit associated with this fix? It would be good to link commits in comments that say "fixed" or "partially fixed" so that other people can follow more easily.

jamesrussell216, 2024-08-14:
@Brow71189 this one I think ^^

Brow71189, 2024-08-14:
Thanks @jamesrussell216. Also I'm fine with closing this since the minimize bug looks fixed.

closed 2024-08-14 by Brow71189