nion-software/nionswift #958: Having Graphics Events would be nice

kociak, 2023-06-14:
This is not a real issue, but I think one thing that is missing atm, to my knowledge at least, are events related to the `Graphics`. At the moment, we have the `graphic_selection_changed_event` in the `DisplayItem` class. However, it only tells you when the selection changed, but not when the `Graphics` attributes are changed. 
My problem is that I can't get notified for example when an `Interval` `start` attribute is changed (when the user is for example moving it)
I dived into the `InspectorPanel`to see how this was handle, but it is a bit cumbersome.
The other way is to make a `computation` but from memory the API is not stable atm. Also, `computation` may be a bit too much for simple applications.
Am I missing something, or maybe it is far on the roadmap because other users do not need it?

cmeyer, 2023-06-14:
Observing a property on a graphic is relatively easy using `Stream.PropertyChangedEventStream`. However, it takes extra (extreme?) care to ensure that everything is handled properly when the user deletes the graphic.

Can you describe your application? Maybe this would be a good example for me to put together... for instance, I could see adding an interval to a line plot, attaching a `Stream.PropertyChangedEventStream` to it, and updating another graphic with the mean value of the data inside the interval. An example could show some of the pitfalls of long calculations in the UI thread, handling deletes, and maybe even undo or persistence. Overall this is a target use case for us, but we're not there yet!

kociak, 2023-06-15:
I see. Of course, the main issue you have here is the one solved by Computations (making sure the listener will not do tons of computations inside the UI thread).
So I think most of the applications could be solved by computations, typically using graphics to select regions of a 1D/2D/ND `data_item`to perform an operation on it.
At the time being, I am unsure how to do it correctly by end (ie, editing the computation panel, but then i modify the original computation?).
Also, nionswift coders may certainly want to do it programmatically (I have one example where I am creating one workspace per spectrum image, then put the spectrum image, a pick tool, a reduce operation and there results in the workspace), but I know the API is not so stable, and it is also not very easy to do.
TBH, i can't see examples where computations will not do the job (except maybe for display performances reasons) - these are really amazing tools. But as I said, i find the computations difficult to set up programmatically and don't use them much.

cmeyer, 2025-03-20:
Closing this since it has no specific action. Documentation for computations can be (partially) tracked here:
- https://github.com/nion-software/nionswift/issues/1366

closed 2025-03-20 by cmeyer