nion-software/nionswift #960: Locked graphics should visually indicate that they are locked

cmeyer, 2023-06-20:
See also:
- #105

jamesrussell216, 2024-07-16:
@cmeyer how do you lock a graphic??

cmeyer, 2024-07-16:
In the inspector

[image]

jamesrussell216, 2025-03-26:
How do we want to visually indicate this @cmeyer ? Change the colour of the "little green squares at the corners and one at the top for rotation that James can't think of the name of right now"? We'll need to be mindful that Luke's graphics PR is going to allow graphics to change colour, but that doesn't extend to those [data cursor things].

jamesrussell216, 2025-03-27:
Discussed this one today. Agreed on...

- The handles changing to crosses when shape is locked.
- The central "+" changing to a cross when position is locked.

Ion-e, 2025-03-28:
Duplicated closed issue: https://github.com/nion-software/nionswift/issues/1331

cmeyer, 2025-04-14:
After trying out the changes in PR #1424, I'm very negative on using 'x' for an endpoint. It looks non-standard and not intuitive. it conveys "danger" to me rather than something as benign as being locked. I think we should abandon that idea and explore alternatives, probably just a square with white interior, or a separate lock attached to the graphic.

cmeyer, 2025-05-28:
PR
- #1424

closed 2025-05-28 by cmeyer

Ion-e, 2025-06-10:
This ticket (and https://github.com/nion-software/nionswift/pull/1424) seem very confusing. Could I get the final specs of what was introduced as bullet points so that I can create a set of tests for it?

jamesrussell216, 2025-06-13:
Review this one after:

https://github.com/nion-software/nionswift/issues/1535
https://github.com/nion-software/nionswift/issues/1532