nion-software/nionswift #961: Line-type graphics should resize from center when position locked or option-key is down

cmeyer, 2023-06-20:
See also:
- #105

Ion-e, 2024-09-05:
@cmeyer Sorry could you elaborate on which graphics do you consider "line-type", and what do you mean by "center"? :)

Ion-e, 2024-09-12:
Answer: normally graphics are resized from the opposite end of the drag

cmeyer, 2024-11-21:
For rectangle-likes:

On macOS it's just 'option' to keep center constant (or the reverse in case of ellipse) and the 'shift' key keeps it square.

On Linux/Windows it's 'alt' to keep center constant (or the reverse in case of ellipse) and 'shift' to keep it square.

I propose to make `.alt` keep the line centered.

The `.alt` modifier maps to 'option' on macOS and 'alt' on Windows. So it's the proper method to use.

The logic for the rectangle is here: https://github.com/nion-software/nionswift/blob/d5f747ba6382792ed5d7e4e2e435fdb69b7037fe/nion/swift/model/Graphics.py#L204

closed 2025-01-13 by cmeyer

Ion-e, 2025-01-20:
Verified as implemented. Thanks!

Closed as per comment: https://github.com/nion-software/nionswift/issues/1311#issuecomment-2598752676

commit 3f900f93 (2025-01-13): Fix #961. Added Alt/Opt to lines to extend from center. Tests.