nion-software/nionswift #983: Using shift for multi-select panels conflicts with rectangle graphic square constraint key

cmeyer, 2023-10-30:
From email

> If I select window A, and then want to take an FFT of a region in window B.
> I choose rectangle tool, draw a rectangle in window B-holding down shift to make the FFT have isotropic pixels. Then do ctrl-F. Nothing will happen because when I drew the square I needed the hotkey shift, but the shift hotkey is also overloaded as multiselect.
> Note that the click in B is not shift click but shift unclick.
> 
> Correct behavior should be that multiselect of windows should happen when the second window is chosen by shift click, but NOT if a tool is used on the window that takes a shift modifier and the activation of the window is a secondary effect of using the tool.
>

Tiomat85, 2024-07-17:
Obvious first question here would be is are we trying to solve the right problem here?

Shift is a modifier for clicking onto a Window (multi-select).
Shift is also a modifier for using the rectangle tool.

So the problem is an ambiguity on what 'Shift' means when a user clicks on another window with the rectangle selected.  In the scenario mentioned the 'expected behaviour' is to ignore the multi-select condition, but follow the rectangle one. Is this always going to be the case though, could the user have clicked the window to indeed attempt to multi-select and have simply forgotten they have the rectangle tool selected which would lead to confusion?

The proposed solution of only accepting the 'Window level' modifiers when using the Pointer tool (as tooltip says, used for selecting display items) makes sense so I am just asking this to make sure there isn't an underlying issue that wants to be discussed.

cmeyer, 2024-07-22:
I don't think there is a significant underlying issue - I think the proposed solution would be reasonable and would not surprise the user. Of course, when you go to implement a solution, it may be more difficult than expected for some reason, in which case we might want to rethink things.

ndellby, 2024-07-22:
There is no ambiguity in the first place, the rectangle tool only works if you shift + click + drag, which makes no sense for selecting windows.
 I can see some minor irritation coming from multiselect not working if I have say the hand (cntrl+H) selected, and that a slightly smoother implementation would ignore multiselect if the selected tool already handles shift as a modifier; that being the conventional way of handling messages in a hierarchy; if you don't handle it but if you handle it then you don't pass it on.

commit e29571ca (2024-08-14): Fix #983. Bypass panel multi-select when only shift and tool utilizes shift.
closed 2024-08-14 by cmeyer