nion-software/nionswift #997: Record abnormal exits from the program and a way to report them

cmeyer, 2024-01-05:
See also
- #977

Swift could record abnormal restarts of the program (didn't exit cleanly). Combined with recording the activity stream, a UI mechanism to generate a report, and perhaps a way to ask for the exit status would allow an improvement in quality.

KRLango, 2024-12-10:
Some notes from our experience of using logs for similar purposes:

We had a try-catch at the very top level of the application that would catch any uncaught exceptions on the main thread and save a crash report with the error message and stack trace. We also added a "thread manager" to route any uncaught exceptions on threads to the same function. We also had logs for starting the software.

The other thing we found helpful to log was what the user was doing. There was a balance here though in that you don't want to log events at such a low level that you are generating multiple logs per interaction, but also you want enough to see what the user was doing. I think you mentioned something about logging actions that have been triggered. That sounds like a good balance to me, we had something similar. For us the actions were often minutes long, if not hours, so we logged start and end of the action, and also we caught and logged any exceptions within the action, finally we also logged when an action was added to the queue. This covered all bases for us, and was very helpful in troubleshooting problems where users did not record any information about an exception at the time. We also logged additional information on a more ad-hoc basis where we found particular things that were useful to us, for example any hardware problems such as if we lost comms with a camera.

jamesrussell216, 2024-12-12:
Thanks @KRLango

cmeyer, 2025-02-19:
PR https://github.com/nion-software/nionswift/pull/1342